PX4/PX4-Autopilot #6431: PlayUAVosd: PX4 problems.

tubeme, 2017-01-24:
Hello,

I reported this in https://github.com/mavlink/qgroundcontrol/issues/4449

And then Lorenz did this patch: https://github.com/PX4/Firmware/commit/1c42cea28e6e09c75ed2c8ab5c012ee4465f9ffb

But I flashed today the Master on the PX4 and connected the PlayUAVosd and again the same problem as in #4449

LorenzMeier, 2017-01-25:
Its impossible that you get this very error message since I commented it out. Can you please be more specific? You have for sure not the "same problem". What is your new problem?

tubeme, 2017-01-25:
The moment I connect the PX4 to QGC even before the parameters are loaded I get the same exact screens as in: mavlink/qgroundcontrol#4449 Exactly the same screens. I can make a full LOG with DebugView but I allready provided log in  mavlink/qgroundcontrol#4449.

LorenzMeier, 2017-01-28:
Please re-flash master. The build server was silently broken last week and its possible you never actually got that fix.

tubeme, 2017-01-28:
ok I will do this.

tubeme, 2017-01-31:
@LorenzMeier Solved! It gives once the WPM: REJ. CMD: Req. WP unexpected error but when closed it is perfectly fine. I will try to make them correct their code. BTW this OSD can load and show the mission (graphically) while in flight so probably there is the problem. It requests the mission from the PX4 in some inappropriate way.

Thanks Lorenz!

closed 2017-01-31 by tubeme