PX4/PX4-Autopilot #6432: Joystick: Not able to arm with RC

tubeme, 2017-01-24:
Hello,

We've had some problems with the Joystick with the new version in 
https://github.com/mavlink/qgroundcontrol/issues/4442

But @DonLakeFlyer and @NaterGator help resolve the issue.

But now there is another set of problems.

It does not matter if the COM_RC_IN_MODE is 1 or 2. all the time  the joystick and RC are fighting. All the control surfaces shake on button arming. And I cannot arm with command, because the rudder is shaking so cannot hold enough time to arm.

https://youtu.be/cnuI7a1QrIE

If I choose mode 1 I understand that the RC is totally in control, except if the RC is switched off. 

I was able to turn off the QGC, arm, then turn on the qgc. In this very moment strange things happen. All the control surfaces shake and this time the throttle is pulsing as well as expected. I was able to turn off the RC in this configuration. Then even stranger thing happened. The throttle gave about 20% constant throttle. But the Joystick was able to control the surfaces this time and the joystick throttle was working.

https://youtu.be/_8KDT3s7jM4

Really strange behavior. We had to turn off the joystick in order to be able to fly.

I suggested that it could be much more easier if there is a software switch in QGC that we can transfer the control from the rc to the joystick and vice versa.

dagar, 2017-01-24:
This is partially fixed here - https://github.com/PX4/Firmware/pull/6391
I'll give it another test and merge later tonight if it works as expected.

LorenzMeier, 2017-01-28:
I consider this fixed in master. This doesn't mean that we have implemented handover between joystick and RC, but at least you now get consistent behaviour. Someone wanting to use both at the same time will need to put up the time to architect the whole handover logic and API on the ground control side. That's probably worth a few days implementation and testing and unless there is a generic need for it not something we have bandwidth to pick up.

closed 2017-01-28 by LorenzMeier