PX4/PX4-Autopilot #6434: position control tilt increase from upward flight to 0

Stifael, 2017-01-25:
When flying position control, the vehicle show a tilt increase when flying upward to hover.

Stifael, 2017-01-25:
a sitl log:
http://logs.px4.io/plot_app?log=e4df0269-6a8b-482c-a169-a5fe384a9d59
From the log it looks like the switch from velocity setpoint control to position control seems to be abrupt with a step in position setpoint. This abrupt switch with not correct position setpoint reset causes a very low thrust setpoint and as a consequence causes a roll setpoint.
[image]
[image]

LorenzMeier, 2017-01-26:
It really looks like a transformation error when going from 3D acceleration to attitude.

Stifael, 2017-01-26:
I will go step by step through the math where the mapping occurs from thrust_sp to attitude_sp

Stifael, 2017-01-26:
The math seems to be correct, at least the part which are used during this maneuver. I also tested what happens when I set the vehicle into an orientation with x = [1,0,0], y = [0,1,0], and z =[0,0,1] and just wen upward. in this scenario, the vehicle should then always try to keep that orientation, and that is also what happens.

Stifael, 2017-01-26:
Ok, setting the attitude setpoint to [1,0,0,0] then it works. so it is definitely a mapping issue. I am going ahead and change that part to quaternion mapping since it is lees prone to errors

Stifael, 2017-01-26:
I recomputed the desired attitude from thrust_sp with quaternions and dont consider yaw at all (the section here: https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L1948-L2003), and I still get the same output. this means it happens farther up the code when the thrust setpoint is computed.

Stifael, 2017-01-26:
current state:
I commented out almost everything except for the velocity controller to reduce the possible causes. It turns out that at low thrust_sp in z direction (for instance at thrust_min_z), the velocity error in x and y produce  desired thrust_sp in x and y which are relative to the z direction large and therefore become more critical. One way to verify the above analysis is to set thrust_sp in x and y to zero or increase the thrust_min_z.

LorenzMeier, 2017-01-26:
Thanks for the update - excellent communication with the rest of the dev team!

Does that mean X and Y need to be scaled to be consistent with hovering throttle plus side throttle? What I mean with that is that the X and Y demands assume keeping the altitude and so the total Z direction should be compensating gravity plus the horizontal demand. If the system is not keeping Z constant then the total thrust vector would need to be scaled.

closed 2017-04-10 by Stifael