PX4/PX4-Autopilot #6436: latest master: region `flash' overflowed by 304 bytes

ChristophTobler, 2017-01-25:
When running ```make px4fmu-v2_default```, there is an overlow

```
[100%] Building C object src/firmware/nuttx/CMakeFiles/firmware_nuttx.dir/builtin_commands.c.obj
[100%] Linking CXX executable firmware_nuttx
/home/christoph/gcc-arm-none-eabi-5_4-2016q2/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: firmware_nuttx section `.data' will not fit in region `flash'
/home/christoph/gcc-arm-none-eabi-5_4-2016q2/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: region `flash' overflowed by 304 bytes
collect2: error: ld returned 1 exit status
src/firmware/nuttx/CMakeFiles/firmware_nuttx.dir/build.make:176: recipe for target 'src/firmware/nuttx/firmware_nuttx' failed
make[3]: *** [src/firmware/nuttx/firmware_nuttx] Error 1
CMakeFiles/Makefile2:6644: recipe for target 'src/firmware/nuttx/CMakeFiles/firmware_nuttx.dir/all' failed
make[2]: *** [src/firmware/nuttx/CMakeFiles/firmware_nuttx.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make[1]: *** [all] Error 2
Makefile:145: recipe for target 'px4fmu-v2_default' failed
make: *** [px4fmu-v2_default] Error 2
```

davids5, 2017-01-25:
@ChristophTobler I have seen intermittent issues with this of late on local build . 

```
GIT_SUBMODULES_ARE_EVIL is defined - Skipping All submodule checking!
-- PX4 VERSION: v1.5.2-893-g63057d7
-- CONFIG: nuttx-px4fmu-v2-default
-- CMAKE_MODULE_PATH: /home/david_s5/Desktop/dev/PX4/repos/mainline/Firmware/cmake
-- Nuttx build for px4fmu-v2 on m4 hardware, using nsh with ROMFS on px4fmu_common
-- The ASM compiler identification is GNU
-- Found assembler: /usr/lib64/ccache/arm-none-eabi-gcc
-- Found PythonInterp: /usr/bin/python (found version "2.7.5") 
-- C compiler: arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]
-- C++ compiler: arm-none-eabi-g++ (GNU Tools for ARM Embedded Processors) 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]

```
Did you do a ```make clean``` and is it repeatable?

ChristophTobler, 2017-01-25:
@davids5 I ran ```make px4fmu-v2_default clean``` and it was repeatable on my desktop.

ChristophTobler, 2017-01-26:
@davids5 I tried on my notebook and there it worked. I also tried ```make clean``` on my desktop and got the same output. ```arm-none-eabi-gcc --version``` produced the same output on both devices:
```
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

davids5, 2017-01-26:
@ChristophTobler 

This is odd! Is it master with no changes?

Please try the following:

```ccache -C```
```git submodule sync --recursive && git submodule update --init --recursive```

The rebuild.

The next thing would be to ``` make distclean``` but it can be destructive as it will delete ANY untracked files.  ***use with caution***

```
distclean: submodulesclean
        @git clean -ff -x -d -e ".project" -e ".cproject" -e ".idea"
```
See https://git-scm.com/docs/git-clean

ChristophTobler, 2017-01-26:
@davids5 I had some changed files which made it trigger... Thank you and sorry...

closed 2017-01-26 by ChristophTobler