PX4/PX4-Autopilot #6443: EKF2 - GPS fusion fault?

pfkrp, 2017-01-26:
Hi guys,
after a few days of testing EKF2 something odd happened. While executing some simple mission, just after take off, my copter started to run away. At first I thought that I made something wrong during mission planning because it seemed as if the copter began heading in the direction of second WP, but it accelerated and passed the WP about 8m from that. I took over control but it was too late and it had hard landing. I have examined the log and don't know what to think of it. In estimator_status there were several GPS glitches reported, problems with velocity and position estimates. No filter_fault_flags, no gps_check_fail_flags. Seems that the estimator 'knew' that something was wrong but did nothing about it. 
A few days later I was testing again. Everything was okay, but after approx 5 flights the same happened. Both logs look the same, there is a huge drift of estimated position. From what I observed I can say that GPS position and GPS velocity (up to 12m/s despite MPC_XY_VEL_MAX = 1m/s) where correct. 
The issue seems to be similar to [#6388](https://github.com/PX4/Firmware/issues/6388) but in real flights, not in SITL. 
I cannot upload whole logs due to some agreements that I'm involved in but I'd just like to report the strange behaviour and release the brain storm.

**FIRST FLIGHT**
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

**SECOND FLIGHT**
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

tumbili, 2017-01-27:
@pfkrp Could you please add plots of the estimator innovations?
It's this topic https://github.com/PX4/Firmware/blob/master/msg/ekf2_innovations.msg

bkueng, 2017-01-27:
Here's a recent log from our flight test people where something similar happened: http://logs.px4.io/plot_app?log=4fa1b2b4-8b19-4f77-815b-81fb8855834c

priseborough, 2017-01-27:
I can't do ekf analysis based on screen shots, please provide the log file.

pfkrp, 2017-01-27:
@tumbili @priseborough Could you please email me on pfkrp0@gmail.com ?

pkocmoud, 2017-01-27:
While bench testing sitting still with GPS connected a Pixracer several days ago, I watched the ship icon drift considerably in a straight line on the QGC map. After a couple seconds, It jumped instantly back to where it started and I saw "EKF GPS fusion timeout - reset to GPS" warning. Not sure it is related but I thought it was odd. If you think this is related, I can spend more time seeking this condition?

tumbili, 2017-01-28:
@pkocmoud You wouldn't have a log of that would you?

Antiheavy, 2017-01-29:
I've seen a similar issue on several occasions and marked it up to bad GPS.  All I can provide is a few anecdotes that may or may-not be helpful:

1) Had an aircraft with a Pixracer and a very-very cheap "ublox m8n" (likely a knock-off) GPS/Mag.  On several occasions the aircraft position on the Qgc map would jump around and looked a lot like the first x-y plot shown at the top of this thread.  It "seemed" like it had a very good GPS position with lots of sats. This same aircraft had flown dozens of times previously without issue.  After a couple outings where this was happening the cheap GPS was replaced with a much better MRO brand M8N unit and never saw the issue again on that aircraft (replaced the wires to the GPS too).

2) Had a different aircraft, again with a Pixracer and MRO M8N GPS.  Aircraft had previously flow great then one day starting behaving very badly with position and heading jumping all over the place.  Discovered that the wires going to the GPS/Mag had been damaged and may have been shorting on a metal screw.  Not sure if the GPS or Mag wires were the ones shorting or what was going on exactly.  Replaced the wires and haven't seen the issue again on that aircraft.

3) I've seen drift like this when set up on a bench inside and I have a very poor in-and-out GPS lock.

brdgtannj, 2017-02-15:
@pfkrp I've been noticing the same issues!  What version of the PX4 firmware have you been using?  Mine only started doing this recently, so I'm wondering if it had something to do with when I merged the newest commits from Master.  I'm going to do a couple of builds to test this out, but my last big update was from Jan. 14th.

pfkrp, 2017-02-16:
@brdgtannj Check how your magnitude of magnetic field behaves. It seemed that it was my problem when I compared to the other flights that were successful. It was caused by grounded metal shield in the vicinity of the gps module. Since removal of the shield I haven't observed that problem and the magnitude of magnetic field looks flat as it should.

LorenzMeier, 2017-05-01:
We've changed so much (and improved so much) that I expect this to be solved on a correctly calibrated setup. Please re-open if it persists.

closed 2017-05-01 by LorenzMeier

tstellanova, 2017-07-11:
Seeing something similar flying an S1000 octo:  
http://logs.px4.io/plot_app?log=d6944989-20ef-4007-bf1e-ca0b3732347f

Multiple warnings:
`2:24:865 | WARNING | [lib__ecl] EKF GPS fusion timeout - reset to GPS`

Have tried two different GPS units (ublox M8N, M8T) with the same results.
Looking at the code it appears this warning shows up when EKF decides GPS data is now valid. I'll look at whether GPS data stalled somehow, but it should not have.

LorenzMeier, 2017-07-11:
It looks more like an inertial sensor issue / EKF2 tuning issue for your specific airframe. You have a relatively high angular rate noise: http://logs.px4.io/plot_app?log=d6944989-20ef-4007-bf1e-ca0b3732347f#Nav-Raw-Angular-Speed-Gyroscope

dagar, 2017-07-11:
Output from `./Tools/ecl_ekf/process_logdata_ekf.py`

[Uploading d6944989-20ef-4007-bf1e-ca0b3732347f.ulg.pdf…]()
It looks like the compass is at least part of the problem.

[image]