PX4/PX4-Autopilot #6444: RTL allows user take over without mode switch

lovettchris, 2017-01-26:
RTL kicked in (set to 50 meter radius), and it brought the drone home and during landing user wanted to nudge the x-y position a bit, but as soon as they touched the sticks the drone immediately switched to stabilize mode, and drone started to plummet until user realized and ramped up throttle to catch the free fall.  I love the idea of being able to nudge the drone while landing (like Solo) but it would be nice if this didn't turn off auto-landing.  This was using latest PX4 master bits from 1/25/2017 with ekf2.

See [log description ](http://www.lovettsoftware.com/Downloads/Bugs/RTLIssue.png)and [actual PX4 log](http://logs.uaventure.com/view/6MLvpYmTqh7N5YerRHjRug).

dagar, 2017-01-27:
So the intention is that after a geofence breach you should be able to easily continue your flight after RTL brings your vehicle back within the boundary.

Options that immediately come to mind.

 1. only allow mode revert to happen for some defined time after the geofence breach (eg 10s)
 2. RTL land actually initiates LAND mode. The mode change would invalidate the possibility of geofence escape and indicate to the user that that you're landing.

As far as tweaking the landing via stick input, that type of thing has been debated off and on. I personally like the idea of any AUTO mode ignoring user input. If you want to land like this why not switch to position mode instead?

lovettchris, 2017-01-27:
Fair enough, my assumption was RTL was a type of auto mode, so I was surprised by the difference in behavior here.  The 3DR solo allows x-y tweaks during landing - it's a nice way to ensure you land right on the launch pad, which is handy in rough terrain.  I would be fine with RTL working like other auto modes requiring mode switch to take control - as it is a bit too easy to bump the sticks accidentally and not realize what you did until the drone hits the ground...  I realize one can learn these things, but the question then is what will new users expect...

AndreasAntener, 2017-01-27:
We too noticed during testing that this is quite confusing. Mainly because the user doesn't necessarily realize that it went into RTL. @dagar option 1) could help there, at least you would see that it's doing something :)
But, I would also be fine with disabling this, needing to flick a switch forth and back seems like a good alternative.
I wouldn't switch to land for the position shift, but rather reuse the implementation for both.

dagar, 2017-01-28:
@AndreasAntener for 1) what do you mean "at least you would see that it's doing something"? Right now the mode does change and is announced in a default QGC setup. There are also mavlink critical messages.

For land I'm talking about the land phase of RTL actually switching to the land navigator state, but that's kind of a separate discussion. 

If this is causing confusion (even among experienced users) I'm not opposed to deleting it.

tubeme, 2017-01-28:
I think the problem is the drastic difference in the modes between auto and manual. If it switches to Position mode then there should be no problems given now you can land in Postition mode.

dagar, 2017-01-28:
The use case imagined was a casual flyer or FPV racer flying and blowing past the geofence, brought safely back within the fence via RTL, and then wanting to easily regain control by wiggling the sticks to resume whatever mode they were in previously. 
Now I'm thinking a mode change to break out is likely good enough and less confusing.

AndreasAntener, 2017-01-29:
@AndreasAntener for 1) what do you mean "at least you would see that it's doing something"? Right now the mode does change and is announced in a default QGC setup. There are also mavlink critical messages.
> doesn't work if you don't have your GCS with you

For land I'm talking about the land phase of RTL actually switching to the land navigator state, but that's kind of a separate discussion.
> yes I know but I think this would make the state switching in commander more complex

dagar, 2017-01-29:
Ah, I keep forgetting that people actually fly without a GCS sometimes. Good point.

lovettchris, 2017-01-30:
It would be safer if the stick bump switched to POS CTRL.  I agree user can learn it is doing this from the heartbeat and/or statustext.  The reason I filed this bug is I was mostly worried about crashing the drone accidentally by bumping the sticks during RTL when no other auto-mode will behave that way.  POS CTRL should solve that.  But it begs the question, why not allow this take over on any auto mode for consistency.  Conversely I would argue if it is not good for every auto mode then why is it good for RTL?   I would still like to nudge on landings, but I'll file that as a separate issue.

dagar, 2017-01-30:
I'm now thinking about getting rid of this behaviour altogether, but I'll try to explain what's currently happening.
It's not allowing the user to regain control in any RTL, it allows a user that was in a manual mode (manual, altitude, position, acro, etc) and breached a geofence to quickly retake control and go straight back into the mode they were in. So this only happens if you have geofence action set to RTL, or LOITER/HOLD and you're in that mode as a result of a geofence breach.

LorenzMeier, 2017-01-30:
I'm thinking in an opposite direction: Rather than trying to get rid of it to switch to the closest mode available with manual input, which is POSCTRL. This makes sense for planes and multicopters and VTOL.

Wanting to adjust / abort the landing can make sense in a lot of situations. But rather than having two pilots, allowing the human to take over is much more consistent.

lovettchris, 2017-01-30:
Speaking of two pilots, see my [new feature request](https://github.com/PX4/Firmware/issues/6483).

Daniel's explanation makes sense, I can see how that is useful if you expect that behavior.  So it allows take over as soon as it enters the fence?  could have some interesting battles between manual and RTL right over that boundary then :-)   

So here's where it was confusing.  While still outside geofence, sticks did nothing.  So one thinks they will continue do nothing until RTL is finished (or you do a mode switch).  How does one know from a long distance away it has re-entered the fence anyway?  (Perhaps there's a status message to this effect? Perhaps we should announce the user can now resume flying or something like that)...

But I could just as easily flip a mode switch to get control back...like I can do with other autonomous modes already.  I can see this could be a religious argument.  Just thought I'd bring it up since we nearly crashed our drone not realizing this was happening.  Flipping to pos control would be safer.

dagar, 2017-01-30:
The geofence action won't trigger faster than once every 10 seconds. So if you violate the fence it will immediately switch to RTL (depending on GF_ACTION), you could then retake control right away and stay outside of the fence, but then 10 seconds later it would retrigger again.

To summarize, could I have your feedback on a specific solution here? @lovettchris @LorenzMeier @AndreasAntener @tubeme 

 - keep geofence failsafe escape functionality via wiggling sticks, but with changes...
   - only escape to position control mode, anything else could be dangerous
   - better user feedback
   - documentation! (QGC setup screen, and guide)
   - possible timeout? If RTL triggers as a result you can only retake control via stick movement for the next < 10 seconds before it's just regular RTL

lovettchris, 2017-01-30:
Looks good, timeout not needed, main thing that resolves this for me is position control.  What about the Landing phase of RTL?  Would this also apply there, or is landing different? I suspect not because I believe it was during landing that we ran into this issue.

LorenzMeier, 2017-01-30:
I would generally allow RTL to be exited to position control if you move the sticks. That seems intuitive.

dagar, 2017-01-30:
Sounds good to me, and it also simplifies the code and user expectation.

We should do a little testing and make sure it's a definite intentional (sustained?) stick wiggle that makes sense on both RC and gamepad/joystick.

AndreasAntener, 2017-01-30:
Still not a fan. The suggested improvements all sound fine, but I have the same concern as before. One idea to improve feedback of what's going on would be to let RTL turn the plane towards home first before it lets the user "wiggle" out of it. I imagine this would make sense in general for the GF violation.

dagar, 2017-01-30:
No problem, I'd rather have these discussions now instead of after the code's been written and tested.

What's your preferred solution?

So now we've generalized RTL as something you can break out of by wiggling the sticks which would leave you in posctl mode. Should we also have a param to enable/disable?

lovettchris, 2017-01-31:
Well I vote for the position control fix.  I assume Andreas is probably voting for physical mode switch since he said earlier "needing to flick a switch forth and back seems like a good alternative."  I'm fine with either fix actually.

AndreasAntener, 2017-02-01:
Because the break out is already there we should try to improve it first and see if it gets accepted. We can always make an option to disable it.

dagar, 2017-02-06:
@LorenzMeier is working on a fix here - https://github.com/PX4/Firmware/pull/6523

dagar, 2017-04-15:
Should be fixed by https://github.com/PX4/Firmware/pull/7025.

closed 2017-04-15 by dagar