PX4/PX4-Autopilot #6445: Mag wont calibrate successfully/Excessive offsets

edsellar, 2017-01-27:
I read through 3711 and 3997 before posting.

The issue is that I can't get mag0 to calibrate successfully. I have tested stable/master and a few past versions. All the same, excessive offsets.

Hardware is Pix1 and external 5983.

I thought this may be a hardware issue but I pulled out the pix and it calibrates fine. So then I thought it was the ext compass so I switched it to a 5883 I've used before. Same issue. I then did this with another Pix1 and a 3DR Pix1. All the same result. Also, I tested it with the GPS unplugged/plugged in and the same thing. Even switched screws around that area to titanium to further reduct anything ferrous.

What steps can I take to provide meaningful info to solve this? It is a new issue I haven't seen before but the probability of 3x Pixhawks and 2x Ext Compass means it is likely not software based but wiring based (of which I redid to be sure is correct). How would one isolate this issue?

No matter what I do I get sensors need inspection. If I load in parameters from another plane that are good and I sort of fudge is but them Im not truly calibrated.

Actual Offsets are:

1	1	CAL_MAG0_ID	                      73225
1	1	CAL_MAG0_ROT	                     -1	
1	1	CAL_MAG0_XOFF                      0.020
1	1	CAL_MAG0_XSCALE	              1.000	
1	1	CAL_MAG0_YOFF	                      0.072
1	1	CAL_MAG0_YSCALE	              1.000	
1	1	CAL_MAG0_ZOFF	                      0.777	
1	1	CAL_MAG0_ZSCALE	              1.000	
1	1	CAL_MAG1_ID	                      131594	
1	1	CAL_MAG1_ROT	                      -1	
1	1	CAL_MAG2_ID	                       0
1	1	CAL_MAG2_ROT	                      -1	
1	1	CAL_MAG_PRIME	                       73225	

ANOTHER TEST - 

AL_MAG0_ID	0	6
1	1	CAL_MAG0_ROT	                       0
1	1	CAL_MAG0_XOFF                      -0.047
1	1	CAL_MAG0_XSCALE	               1.000
1	1	CAL_MAG0_YOFF	                       0.058
1	1	CAL_MAG0_YSCALE	               1.000
1	1	CAL_MAG0_ZOFF	                       0.482
1	1	CAL_MAG0_ZSCALE	               1.000
1	1	CAL_MAG1_ID	                       0
1	1	CAL_MAG1_ROT	                      -1	
1	1	CAL_MAG1_XOFF 	              -0.036	
1	1	CAL_MAG1_XSCALE	              1.000	
1	1	CAL_MAG1_YOFF                      -0.098	
1	1	CAL_MAG1_YSCALE	              1.000	
1	1	CAL_MAG1_ZOFF	                      0.001	
1	1	CAL_MAG1_ZSCALE	              1.000	
1	1	CAL_MAG2_ID                             0	
1	1	CAL_MAG2_ROT	                     -1
1	1	CAL_MAG_PRIME	                      73225	

Pix plugged into computer with no ext compass:

1	1	CAL_MAG0_ID	                     131594	
1	1	CAL_MAG0_ROT	                      -1	
1	1	CAL_MAG0_XOFF                      -0.016	
1	1	CAL_MAG0_XSCALE	             1.000	
1	1	CAL_MAG0_YOFF                     -0.086	
1	1	CAL_MAG0_YSCALE	             1.000	
1	1	CAL_MAG0_ZOFF                     0.011	
1	1	CAL_MAG0_ZSCALE	             1.000	
1	1	CAL_MAG1_ID	                     0	
1	1	CAL_MAG1_ROT                       -1	
1	1	CAL_MAG2_ID	                     0
1	1	CAL_MAG2_ROT	                     -1	
1	1	CAL_MAG_PRIME	                      131594	

I reset all parameters and then did a calibration with the ext plugged in:

1	1	CAL_MAG0_ID	                      73225	
1	1	CAL_MAG0_ROT	                      0	
1	1	CAL_MAG1_ID	                      131594	
1	1	CAL_MAG1_ROT	                      -1	
1	1	CAL_MAG2_ID	                      0	
1	1	CAL_MAG2_ROT	                      -1	
1	1	CAL_MAG_PRIME	                      73225

LorenzMeier, 2017-01-28:
Can you try to calibrate outside of the plane (just a bench setup with Pixhawk and external GPS module?).

Its either your plane / copter or you're calibrating near a metal object or you are not actually spinning the system around in six dimensions.

dagar, 2017-04-11:
@edsellar any update?

LorenzMeier, 2017-04-30:
Closing as stale.

closed 2017-04-30 by LorenzMeier