PX4/PX4-Autopilot #6446: Landing won't quite land

lovettchris, 2017-01-27:
Auto landing on today's master PX4 branch using ekf2 on px4fmu_v2-default is much improved, thanks.  But still one minor issue today.  On a Team Blacksheep Discovery, the drone descended smoothly (a bit slowly actually, 0.3 m/s, goal was 0.5) then leveled off about 8 inches above the ground.  I had to cut the motors for it to actually land.  See last 16 seconds of [this log file](http://logs.uaventure.com/view/y5JeUEuiBxsXMDKM2yEk7h).  Perhaps one solution is if altitude sensors are not declining and you haven't seen the big z-acc landing bump yet then just steadily ramp down the motors.  Notice here that the last few seconds before I cut the motors, the motors were not ramping down.

[image]

LorenzMeier, 2017-01-28:
This PR https://github.com/PX4/Firmware/pull/6362 does fix landing and land detection in a very smooth and reliable way. If it doesn't work for you on that branch then the conclusion would be that your system (min throttle, hover throttle) needs proper setup.

Detecting the landing requires a valid vehicle model, which is provided by the min and hover throttle. We can't something generic without those parameters which is still reliable.

closed 2017-01-28 by LorenzMeier

lovettchris, 2017-01-30:
I will try the new bits.  I assumed the QGC configuration for the Team Blacksheep Discovery was already set correctly.  Is there more I need to do? Can you provide a pointer to wiki page on how to configure min/hover throttle correctly ?

LorenzMeier, 2017-01-30:
It will likely require tuning. Its also specific to your build / battery / payload

lovettchris, 2017-01-30:
I see, I wonder if we could do some auto-tuning, like this: [Automatic Learning of Hover Throttle](http://ardupilot.org/copter/docs/ac_throttlemid.html#automatic-learning-of-hover-throttle) ?