PX4/PX4-Autopilot #6449: Pixracer: duplicated driver instances

bkueng, 2017-01-27:
On a recent Pixracer (I don't have that problem on an older one), starting the sensors driver `mpu6000 -R 2 -T 20608 start` instantiates two driver objects, resulting in totally 3 sensor publication instances, 2 with the same device id.

Relevant boot output:
```
MPU6000 on SPI bus 1 at 6 (1000 KHz)
MPU6000 on SPI bus 1 at 6 (1000 KHz)
MPU9250 on SPI bus 1 at 4 (1000 KHz)
```

The problem seems to be that several bus options in [0] match, without comparing the device_type.

@davids5 What do you think is the best option to resolve this? adding a device_type to `mpu6000_bus_option`?

@priseborough I think this is the problem you're experiencing

[0] https://github.com/PX4/Firmware/blob/master/src/drivers/mpu6000/mpu6000.cpp#L2227

davids5, 2017-01-27:
@bkueng 

I think a quick fix is to NOT use the default "all" in the startup script for FMUv4 but add the [- bus options](https://github.com/PX4/Firmware/blob/master/src/drivers/mpu6000/mpu6000.cpp#L2626-L2640)

The correct fix is to have boards configs supply device that can be probed for to the drivers.  The is on my list of to-dos.

I do not think device_type would be the correct criteria as a board could have 2 of the same sensors.

The mapping from logical chip select (PX4_SPIDEV_xxx) maps to a single physical line per bus. Once that bus - cs pair is claimed - it can not be used again.

bkueng, 2017-01-27:
Adding a busoption `-s` does not work, because both are marked as MPU6000_BUS_SPI_INTERNAL1

davids5, 2017-01-27:
ahh - I can have a look at it 

When you say recent when did you get it? 
What chips are on your board?

pkocmoud, 2017-01-27:
When I run this command on a fresh Pixracer I get:
mpu6000 -R 2 -T 20608 start
MPU6000 on SPI bus 1 at 6 (1000 KHz)

LorenzMeier, 2017-01-28:
The device ID code is wrong - it likely doesn't set the device ID struct properly

LorenzMeier, 2017-01-28:
Fix in: https://github.com/PX4/Firmware/pull/6461

bkueng, 2017-01-30:
The device ID is a different problem, it's not fixed yet. I'll propose a fix.

PR #6486 (2017-01-31): MPU6000: check against device_type
Fixes the following case: a board that has several bus_options enabled,
and wants to start a specific device (eg mpu6000 -T 20608 start). In that
case the given device_type is never actually checked, and since the start
routine iterates over all bus_options, several can match, and thus start
multiple instances of the driver (note that the whoami check is performed
against the value provided with -T and thus the check will pass).

It happens on Pixracer (PX4_SPIDEV_MPU and PX4_SPIDEV_ICM_20608 are both set).

This is a short-term fix, @davids5 plans to refacto [... cut]
closed 2017-01-31 by LorenzMeier