PX4/PX4-Autopilot #6456: FW auto land with intermittent terrain data

dagar, 2017-01-28:
Fixed wing crash reported by @Flying_Forfun on px4 discuss during auto land with a lidar light. Unfortunately no sdlog is available, which is another bug.

During fixed wing auto land the terrain data becomes intermittant and due to teh confusion the vehicle starts climbing, initiates flare at > 60m and stalls during the portion where the pitch setpoint is forced (https://github.com/PX4/Firmware/blob/master/src/modules/fw_pos_control_l1/fw_pos_control_l1_main.cpp#L1668)

@AndreasAntener @tumbili what do you think?

http://discuss.px4.io/t/cant-switch-out-of-mission-mode-during-a-mission/2294

[image]

LorenzMeier, 2017-01-28:
That whole feature is highly experimental and not application-hardened. It will need some love and whoever picks it up needs a bit of time. We do have a lidar in Gazebo (mounted to the optical flow iris), so the best would be to start with a pull request for sitl_gazebo which adds it to the plane model and starting testing there.

dagar, 2017-02-09:
From a proper log for reference.

[image]

When the altitude setpoint (green) drops to ~190m the vehicle is landing, but again flares early.

The FW position controller should be aborting landing if the the terrain data is reported invalid (not logged), but only if the last update is > 1s. These dropouts are much shorter.

dagar, 2017-02-09:
@LorenzMeier do you think an appropriate fix here would be EKF2 tuning (EKF2_RNG_NOISE, EKF2_RNG_GATE), better handling of terrain data in fw_pos_ctrl, or both?

closed 2017-04-30 by dagar