PX4/PX4-Autopilot #6469: IO Firmware build: Incorrect CMake dependencies

LorenzMeier, 2017-01-29:
@dagar @jgoppert Something with the dependency generation is odd:

```
make px4fmu-v2_default upload
[147/165] Building CXX object src/modules/px4iofirmware/CMakeFiles/px4io-v2.dir/__/systemlib/mixer/mixer.cpp.obj
FAILED: src/modules/px4iofirmware/CMakeFiles/px4io-v2.dir/__/systemlib/mixer/mixer.cpp.obj 
/usr/local/bin/arm-none-eabi-g++   -DCONFIG_ARCH_BOARD_PX4IO_V2 -D__DF_NUTTX -D__PX4_NUTTX -D__STDC_FORMAT_MACROS -I. -Isrc -Isrc/modules -Isrc/modules/px4_messages -I../mavlink/include/mavlink -I../src -I../src/drivers/boards/px4io-v2 -I../src/include -I../src/lib -I../src/lib/DriverFramework/framework/include -I../src/modules -I../src/platforms -I../src/modules/px4iofirmware/src/lib/matrix -Ipx4io-v2/NuttX/nuttx/nuttx-export/include -Ipx4io-v2/NuttX/nuttx/nuttx-export/include/cxx -Ipx4io-v2/NuttX/nuttx/nuttx-export/arch/chip -Ipx4io-v2/NuttX/nuttx/nuttx-export/arch/common -Ipx4io-v2/NuttX/nuttx/nuttx-export/arch/armv7-m -Ipx4io-v2/NuttX/apps/include -Isrc/modules/systemlib/mixer -I../src/modules/px4iofirmware/. -g -fno-exceptions -fno-rtti -std=gnu++0x -fno-threadsafe-statics -DCONFIG_WCHAR_BUILTIN -D__CUSTOM_FILE_IO__ -fcheck-new -Wall -Werror -Wextra -Wno-sign-compare -Wshadow -Wfloat-equal -Wpointer-arith -Wmissing-declarations -Wno-unused-parameter -Werror=format-security -Werror=array-bounds -Wfatal-errors -Werror=unused-variable -Werror=reorder -Werror=uninitialized -Werror=init-self -Werror=unused-but-set-variable -Wformat=1 -Wdouble-promotion -Werror=double-promotion -Wno-missing-field-initializers -fvisibility=hidden -include visibility.h -mcpu=cortex-m3 -mthumb -march=armv7-m -nodefaultlibs -nostdlib -Os -fno-strict-aliasing -fomit-frame-pointer -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -MD -MT src/modules/px4iofirmware/CMakeFiles/px4io-v2.dir/__/systemlib/mixer/mixer.cpp.obj -MF src/modules/px4iofirmware/CMakeFiles/px4io-v2.dir/__/systemlib/mixer/mixer.cpp.obj.d -o src/modules/px4iofirmware/CMakeFiles/px4io-v2.dir/__/systemlib/mixer/mixer.cpp.obj -c /Users/user/src/Firmware/src/modules/systemlib/mixer/mixer.cpp
In file included from /Users/user/src/Firmware/src/modules/systemlib/mixer/mixer.cpp:40:0:
../src/platforms/px4_config.h:44:26: fatal error: nuttx/config.h: No such file or directory
compilation terminated.
```

LorenzMeier, 2017-01-29:
IO is rebuilt on ROMFS changes. That makes no sense at all.

dagar, 2017-01-29:
Something is definitely not right with the NuttX build dependencies. It must be touching something one build which then causes a larger rebuild the next.

	 % ninja -j1
	[48/48] Generating px4fmu-v2_default.px4, firmware_nuttx.bin

	 % ninja -j1
	[2/2] Copying NuttX for px4io-v2 with nsh

	 % ninja -j1
	[48/48] Generating px4fmu-v2_default.px4, firmware_nuttx.bin

	% ninja -j1
	[2/2] Copying NuttX for px4io-v2 with nsh

	% ninja -j1
	[48/48] Generating px4fmu-v2_default.px4, firmware_nuttx.bin

This might be a good excuse to move the cmake nuttx build wrapper to ExternalProject_Add. https://cmake.org/cmake/help/v3.2/module/ExternalProject.html

That's effectively what we've ended up recreating from scratch. https://github.com/PX4/Firmware/blob/master/cmake/nuttx/px4_impl_nuttx.cmake#L259

@davids5 FYI

jgoppert, 2017-01-29:
I originally implemented it with external project, there was some reason at the time it was annoying or wouldn't work.

dagar, 2017-01-29:
I think I found a simple fix. Happy to leave it alone for now. #6475

commit 159b3591 (2017-01-30): cmake nuttx build use full path to copy stamp

 - fixes #6469
closed 2017-01-30 by LorenzMeier