PX4/PX4-Autopilot #6474: Vulnerability in px4_log.c

vvranjek, 2017-01-29:
I am running a slightly modified version of v1.5.3 and I realized that `warnx()` caused a system crash. After some debugging I found out that in `px4_log.c` in function `void px4_log_modulename()`, `int level` was set to 2, which doesn't exist in `log_level_table[]`. Setting it to `level = 4;` right after `const uint8_t log_level_table[]` fixes the problem. I haven't debugged any further, and will let you guys fix the problem, as you know the code much better then I do.

Let me know if you need any more info.

Cheers

vvranjek, 2017-01-29:
Here is how to reproduce the problem on stock v1.5.3:

Write `warnx("Whatever, testing warnx");' in line 394, file: `tecs.cpp`. 
Run simulator X-Plane 10 in fixedwing. 
Arm, Mission mode
It will crash within a minute.

I know it's a lot of printing, but i tried printing every second and it's the same thing.

LorenzMeier, 2017-01-29:
@vvranjek Thank you for reporting this and spending the time analysing it. I would be interested to understand how you fixed it. Your description above does not hint at an issue to me. Here is how the code will evaluate for a log level of `2`:

Here are the log levels:

```
const uint8_t log_level_table[] = {
			6, /* _PX4_LOG_LEVEL_ALWAYS */
			7, /* _PX4_LOG_LEVEL_DEBUG */
			4, /* _PX4_LOG_LEVEL_WARN */
			3, /* _PX4_LOG_LEVEL_ERROR */
			0  /* _PX4_LOG_LEVEL_PANIC */
		};
```

So using `level = 2` in this call:

```
log_message.severity = log_level_table[level];
```

Will index the *3rd* entry, which is the warning log level = 4. So its equivalent to:

```
log_message.severity = 4;
```

What you did was setting level to 4, which indexes the 5th entry, which is 0 = PANIC, so the resulting code is:

```
log_message.severity = 0;
```

So you basically just raised the severity. I couldn't find anything in the codebase that would reference the severity field incorrectly.

Is your analysis maybe based on a custom module which has a stack overflow or which tries to print messages in interrupt context?

vvranjek, 2017-01-30:
At first I thought that number `2` doesn't exist in the array, but then I realized my mistake. However, setting level to 4 still fixes the problem.

I reproduced the the error on stock v1.5.3, no modifications. Here are some other things that fix the problem:

- Using `printf()` instead of `warnx()`
- Commenting out the whole `if` statement below `/* publish an orb log message */` in px4_log.c.
- Setting level to 4 as described above.

I couldn't find anything that would reference the severity either. Can someone please try to reproduce the problem? I must have tried 30+ times with all kinds of different configurations.

Also, the crash happens without the hardfault message and memory dump. It just freezes until I reboot it.

LorenzMeier, 2017-01-30:
Does the issue also go away if you increase the stack of your module in task_spawn by 200 bytes?

vvranjek, 2017-01-30:
I have set navigator task size to 2000 and it still crashed but it seems later. I realized thought, that as soon as I send `top once` command it crashes at time of printing top or right after. Memory usage seems fine. 

I did the right thing raising the memory in `navigator` task, since the change that causes the crash is in `tecs.cpp`, right?

Also, I think it would be nice to log stack usage of each task, don't you think?

vvranjek, 2017-01-31:
Did some more testing and found out that most likely `fw_pos_ctrl_l1` was running out of memory. I raised the stack size to 2000 from 1700 and it haven't crashed since. Previously it would crash within a minute, sometimes right after takeoff. I think the problem is solved, except that the crash never triggered the core dump, thus I was unable to tell which task crashed and it made it difficult to debug. Maybe you guys can have a look at that?

mhkabir, 2017-01-31:
Sorry, didn't mean to close this. Some sort of a mobile Github fail.

julianoes, 2017-03-08:
Modules exceeding there stack should say so in the console and ulog, see:
https://github.com/PX4/Firmware/blob/ffacc6f64a656f6b992e398b1ed860abd0a4f39b/src/modules/load_mon/load_mon.cpp#L298-L301
Therefore closing this issue.

closed 2017-03-08 by julianoes