PX4/PX4-Autopilot #6481: CONFIG_RAM_SIZE may not be correct way to determine RAM size

davids5, 2017-01-30:
See https://github.com/PX4/Firmware/pull/6451#pullrequestreview-18961944

davids5, 2017-02-04:
@LorenzMeier In thinking about the use of this

Should we consider adding system integrity vault. 

There are certain assets stored in ram that are critical and need to be tested for integrity at runtime.

Assuming we consider the OS+NSH vetted (if not we add CONFIG_BOARD_INITIALIZE and call it earlier from board_initialize),  in board_app_initialize we call system_integrity_init() this will fetch the os structs we want to check for corruption. I.E. the results mem.arena from mallinfo. etc  We save the values in the vault and crc it.

system_integrity_init() - populates the vault from compile time list of assets.
system_integrity_verify() can get assets again and check against the vault  (done in pieces if necessary)
system_integrity_verify_vault() - check the crc of the vault.
system_integrity_verify_item(enum handle, *void value) - checks the value return 0 for OK and -1 not OK

Then from places [like](https://github.com/PX4/Firmware/blob/master/src/modules/load_mon/load_mon.cpp#L227)  we can call system_integrity_verify_item(INTEGRITY_MEM_HEAP, &mem.arena).  If the result is not 0 call system_integrity_verify_vault() if it is OK then the heap is corrupted if it is not things are worse

LorenzMeier, 2017-02-04:
Yep, that makes perfect sense. Can you add that to your list?

closed 2018-02-21 by PX4BuildBot