PX4/PX4-Autopilot #6482: LowPassFilter2p stability?

MaEtUgR, 2017-01-30:
The low pass filter in the math library can easily be ran with a configuration that drives it unstable.

I encountered this issue when filtering RC input and choosing the cutoff frequency above (sample frequency)/2. Everything above is unstable and the value itself is critically stable.

Maybe there should be a check for that inside the filter?

dagar, 2018-01-17:
Should we write testing for the filter?

MaEtUgR, 2018-01-17:
The issue was only to suggest catching provable unstable filter configuration before running it. Don't know how important that is or if people know what they are dealing with.

I think a test would probably be overkill, I think it does exactly what it was designed for. It's just not a lowpass filter usable for every scenario because it's simple type of IIR with the characteristics:
- being able to change the frequencies on the fly
- can go unstable depending on configuration
- can overshoot over the input value also no BIBO stability

We probably need also other "go to" maybe FIR filters which are simple to reconfigure and run on a microcontroller. I see use cases for RC, battery, IMU noise.