PX4/PX4-Autopilot #6483: Request for feature: manual nudging of autonomous flight modes

lovettchris, 2017-01-30:
3DR solo allows nudging of x-y position during landing.  This is handy when you want to hit the launch pad when you are in rough terrain.  Why use auto-landing then?  Because it is usually better than manual landing on the vertical control, same reason I use ALT-HOLD mode most of the time.  

But what if we generalized this and allowed minor nudging of position during all autonomous modes.  One time I crashed a 3dr solo into a big dead tree because the mission waypoint went to close, I would have loved to be able to nudge it while it was on mission to help it dodge the tree.   I'm sure the same could be said for raising altitude on a mission that is about to clip the tip of a tall tree.  Or manually lower the altitude when going under a tree canopy.  

I realize this is a hard problem and we may need to implement some more advanced [shared autonomy](http://www.roboticsproceedings.org/rss11/p32.pdf) research where stick input is used to infer the goal user has so we can safely modify the autonomous mission with new goal in mind, rather than having the sticks fight the autonomous mission flight control directly.  Could be fun...

LorenzMeier, 2017-04-30:
We will add this for landing first and then think about how solving the shared autonomy should eventually work.

closed 2018-02-21 by PX4BuildBot

Gubayura, 2025-04-28:
Please, help me understand what is 'nudge'/'nudging' in the context of the following _Enable wheel controller **nudging** while on the runway_
_**Nudging** is done with yaw stick_