PX4/PX4-Autopilot #6484: Bug in Offboard mode

PeterQFR, 2017-01-31:
Hi, I've been developing to use px4 flight stack in offboard mode. However in setting the velocity values in x, y, and z using mavlink_set_position_target_local_ned_t, I noticed that the inputs were not being reflected when receiving the mavlink_msg_posiiton_target_local_ned.

I identified the bug in src/modules/mavlink_receiver.cpp handle_set_position_target_local_ned() function. 

The type_mask for the mavlink_set_position_target_local_ned is poorly concieved as the masks are sequentially numbered 0x1000, 0x2000, 0x3000, 0x4000, which cover multiple bits the net result is that when in loiter mode 0x3000 the controller thinks that is in Takeoff mode due to the exclusive if/else if statement. 
This was causing an issue in the mc_position_controller where it never left the takeoff type and was zeroing x and y velocity setpoints.

My solution was to change the masks to only handle the individual bits thus be numbered 0x1000, 0x2000, 0x4000, 0x8000. 

This correctly handled the the state changes in offboard mode and the velocity setpoints are correctly set and returned from the FC via mavlink.

I can provide a patch to the code and create a pull request if required.

mhkabir, 2017-01-31:
@PeterQFR Would great if you could create a PR. Thanks!

dagar, 2017-01-31:
That's unfortunate. It looks like some of those bits are covered by the mavlink spec, but not the setpoint types. That code might actually be readable if those were DEFINES.
http://mavlink.org/messages/common#SET_POSITION_TARGET_LOCAL_NED

PeterQFR, 2017-01-31:
@dagar instead of #defines I can include the definition of that part of the typemask in the definition file of mavlink common enums, common.xml. It will be generated in common.h and thus it will be available to whomever needs to use the message? I can also include the other typemasks as the description of which bits to set is confusing as the bit numbering starts at 1 instead of 0.

dagar, 2017-01-31:
Yes! That's definitely the better solution.
Mavlink definitions are in this repo - https://github.com/mavlink/mavlink

AndreasAntener, 2017-02-03:
Standard discussion: https://github.com/mavlink/mavlink/pull/670

commit 08891526 (2017-06-15): Fixes #6484: Bug in Offboard mode incorrect use of type_mask
closed 2017-06-15 by LorenzMeier