PX4/PX4-Autopilot #6493: EKF2 won't takeoff with flow only

jgoppert, 2017-02-01:
On master I"m still not able to do flow missions with EKF2. It won't takeoff. My goal here is to compare performance with LPE and IEKF.

jgoppert, 2017-02-01:
Startup and mission send, then arm.

```bash
NFO  [simulator] Got initial simuation data, running sim..
INFO  [pwm_out_sim] MODE_8PWM
Sleeping for 1 s; (1000000 us).
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 14556 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14557 remote port 14540
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
pxh> INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log
INFO  [logger] Opened log file: rootfs/fs/microsd/log/2017-02-01/17_02_49.ulg
INFO  [mavlink] partner IP: 127.0.0.1
INFO  [tone_alarm] startup
INFO  [lib__ecl] EKF aligned, (range height, IMU buf: 21, OBS buf: 16)
INFO  [tone_alarm] negative
INFO  [tone_alarm] negative
INFO  [tone_alarm] negative
```

jgoppert, 2017-02-01:
The aid mask was incorrectly set, but still not luck:

[image]

jgoppert, 2017-02-01:
Has anyone flown this with optical flow in real life/SITL or is all of this code still in development?

mhkabir, 2017-02-01:
EKF2 has never worked in this setup. Probably needs https://github.com/PX4/Firmware/issues/6347.

jgoppert, 2017-02-01:
OK, thought that was already in, I"ll try to force a takeoff by starting with GPS and then disabling it and seeing what the performance is like.

jgoppert, 2017-02-01:
No luck, logic errors, so none of this has actually been tested before, we need to fix that.

```bash
pxh> ekf2 start
pxh> 
pxh> 
pxh> 
pxh> eINFO  [lib__ecl] EKF aligned, (range height, IMU buf: 21, OBS buf: 16)
kf2 status
WARN  [ekf2] running
WARN  [ekf2] local position OK [NO]
WARN  [ekf2] global position OK [NO]
pxh> INFO  [lib__ecl] EKF GPS checks passed (WGS-84 origin set)
INFO  [lib__ecl] EKF commencing GPS fusion
INFO  [commander] home: 47.3977420, 8.5455936, 488.10
INFO  [tone_alarm] home_set

pxh> ekINFO  [tone_alarm] neutral
pxh> 
pxh> commander takeoff
pxh> INFO  [commander] Taking off
INFO  [tone_alarm] positive
INFO  [commander] home: 47.3977421, 8.5455938, 488.11
INFO  [tone_alarm] arming
INFO  [commander] Takeoff detected
Number of good matches: 1, desired: 20
Number of good matches: 2, desired: 19
Number of good matches: 3, desired: 18
Number of good matches: 5, desired: 17
Number of good matches: 2, desired: 13
Number of good matches: 2, desired: 12
Number of good matches: 4, desired: 10
Number of good matches: 3, desired: 9
Number of good matches: 6, desired: 10
Number of good matches: 2, desired: 5
Number of good matches: 3, desired: 12
Number of good matches: 7, desired: 12
Number of good matches: 6, desired: 7

pxh> gpssim stop
INFO  [gpssim] exiting
pxh> [Wrn] [Publisher.cc:140] Queue limit reached for topic /gazebo/default/iris_opt_flow/gazebo/command/motor_speed, deleting message. This warning is printed only once.
WARN  [navigator] global position timeout
INFO  [tone_alarm] gps_warning
INFO  [commander] no gps
WARN  [commander] failsafe mode on
INFO  [tone_alarm] fast_bat
Number of good matches: 1, desired: 2
```

ChristophTobler, 2017-02-01:
@jgoppert Correct, auto takeoff is currently not working. We need to introduce the origin parameters: https://github.com/PX4/ecl/pull/234#issuecomment-273422616
I can have a look at that with @priseborough

jgoppert, 2017-02-01:
Ok, even when in the air and initialized with gps, if you turn gps it still fails. So more to fix than just takeofff. Probably a publishing logic issue.