PX4/PX4-Autopilot #6498: Pixfalcon Buzz and not works on master

artemskorsky, 2017-02-02:
after uploading master branch to pixfalcon it buzzes several times,
and servos not electrified.downgrading to branches like 1.5.x or 1.4.x
does not solve the issue still buzzes in the downgraded versions,
It used to work well in the stable branch.
ps: rc is not connected

artemskorsky, 2017-02-02:
this problem doesnt happen with pixhawk

dagar, 2017-04-11:
Still an issue? Isn't a pixfalcon identical to a pixhawk (from PX4s perspective)?

dagar, 2017-04-28:
Please reopen if this is still an issue and we can debug. It's especially weird because PX4 doesn't know if it's on a pixhawk or pixfalcon.

closed 2017-04-28 by dagar

zlite, 2017-04-28:
I vaguely remember that there was a request to turn the buzzer off on
Pixfalcon and Pixhawk Mini because it's embedded in the AP and causes
vibration issues.

dagar, 2017-04-28:
You can disable the buzzer by setting CBRK_BUZZER.
https://dev.px4.io/en/advanced/parameter_reference.html#circuit-breaker

I wonder if it's enough of an issue to have it off by default?