PX4/PX4-Autopilot #6501: Incremental build broken when switching branches

bkueng, 2017-02-02:
Sometimes when switching between branches, the incremental build is broken, with the following error:
```
[8/621] Building NuttX for px4fmu-v4 with nsh
FAILED: px4fmu-v4/NuttX/nuttx/nuttx-export/include/nuttx/config.h
cd /home/beat/px4/src/Firmware/build_px4fmu-v4_default && /usr/bin/rm -rf /home/beat/px4/src/Firmware/build_px4fmu-v4_default/px4fmu-v4/NuttX/nuttx/nuttx-export && /usr/bin/make --no-print-directory --quiet -C /home/beat/px4/src/Firmware/build_px4fmu-v4_default/px4fmu-v4/NuttX/nuttx -r CONFIG_ARCH_BOARD=px4fmu-v4 export > nuttx_build.log
In file included from ./stdio/lib_libdtoa.c:47:0,
				 from stdio/lib_libvsprintf.c:236:
/home/beat/px4/src/Firmware/build_px4fmu-v4_default/px4fmu-v4/NuttX/nuttx/include/math.h:52:25: fatal error: arch/math.h: No such file or directory
 #  include <arch/math.h>
						 ^
compilation terminated.
ERROR: arm-none-eabi-gcc failed: 1
	   command: arm-none-eabi-gcc -MT bin/lib_libvsprintf.o  -M -std=gnu99 -Wall -Wextra -Werror -Wdouble-promotion -Wframe-larger-than=1024 -Wlogical-op -Wpacked -Wpointer-arith -Wshadow -Wno-sign-compare -Wno-unused-parameter -Wno-nonnull-compare -Wno-misleading-indentation -Wall -Wextra -Werror -Wdouble-promotion -Wframe-larger-than=1024 -Wlogical-op -Wpacked -Wpointer-arith -Wshadow -Wno-sign-compare -Wno-unused-parameter -Wno-nonnull-compare -Wno-misleading-indentation -Wbad-function-cast -Wmissing-parameter-type -Wnested-externs -Wstrict-prototypes -Wno-bad-function-cast -Wno-cpp -Wno-implicit-function-declaration -Wno-maybe-uninitialized -Wno-missing-field-initializers -Wno-nested-externs -Wno-old-style-declaration -Wno-pointer-sign -Wno-type-limits -Wno-unused-but-set-variable -Wno-unused-function -Wno-unused-label -Wno-unused-variable -Os -fno-strict-aliasing -fno-strength-reduce -fomit-frame-pointer -funsafe-math-optimizations -fno-builtin-printf -ffunction-sections -fdata-sections -g -mcpu=cortex-m4 -mthumb -march=armv7e-m -mfpu=fpv4-sp-d16 -mfloat-abi=hard -I. -isystem /home/beat/px4/src/Firmware/build_px4fmu-v4_default/px4fmu-v4/NuttX/nuttx/include -D__KERNEL__ -pipe -fno-common stdio/lib_libvsprintf.c
make[1]: *** [.depend] Error 1
In file included from ./stdio/lib_libdtoa.c:47:0,
				 from stdio/lib_libvsprintf.c:236:
/home/beat/px4/src/Firmware/build_px4fmu-v4_default/px4fmu-v4/NuttX/nuttx/include/math.h:52:25: fatal error: arch/math.h: No such file or directory
 #  include <arch/math.h>
						 ^
compilation terminated.
make[1]: *** [bin/lib_libvsprintf.o] Error 1
make: *** [libc/libc.a] Error 2
```

It looks like the NuttX patches are not applied when rebuilding. `make clean` does resolve it, but I think this should work incrementally as well.
I never had this problem before the NuttX upgrade.

@dagar Can you have a look?

dagar, 2017-02-02:
Is the NuttX submodule changing in these situations?

Generally I'm a little paranoid and do `make distclean` between branches, but I'm often jumping between branches off of latest master, and branches off of stable. 
We should still make incremental build reliable.

bkueng, 2017-02-02:
> Is the NuttX submodule changing in these situations?

I think so but I haven't explicitly checked that.

Thanks for looking into it.

dagar, 2017-02-08:
It's actually not obvious what's going on here. If you use a build with an IO (eg px4fmu-v3) changing the submodules cause cmake to rerun and the nuttx build gets into a bad state, but only for the FMU.

davids5, 2017-02-08:
Is it possible that the patch's stamps are not fully qualified paths?

PR #7007 (2017-04-18): cmake cleanup nuttx dependency handling
 - closes #6501
 - closes #6820
 - closes #6881
commit 3721fb9d (2017-04-18): cmake cleanup nuttx dependency handling (#7007)

 - closes #6501
 - closes #6820
 - closes #6881
closed 2017-04-18 by dagar