PX4/PX4-Autopilot #6515: Bebop 2 / RPi toolchain update + Skycontroller 2 build system

mhkabir, 2017-02-04:
We should update to the GCC 5.3 toolchain for the Bebop 2 and Raspberry Pi. There is no longer a need to maintain a separate Raspbian-specific toolchain for RPi.

New Bebop 2 / RPi toolchain :
https://releases.linaro.org/components/toolchain/binaries/latest-5/arm-linux-gnueabihf/gcc-linaro-5.3.1-2016.05-x86_64_arm-linux-gnueabihf.tar.xz

Will also be introducing the build system for the Parrot Skycontroller 2, which will be able to run PX4 and act as a generic long-range Mavlink remote controller for the Bebop and any other vehicle with wifi.

Skycontroller 2 toolchain : 
https://releases.linaro.org/components/toolchain/binaries/latest-5/arm-linux-gnueabi/gcc-linaro-5.3.1-2016.05-x86_64_arm-linux-gnueabi.tar.xz

mhkabir, 2017-02-04:
Bebop builds with new toolchain : 

```-- PX4 VERSION: v1.6.0rc1-104-g0490bfe
-- CONFIG: posix-bebop-default
-- CMAKE_INSTALL_PREFIX: /usr
-- CMAKE_MODULE_PATH: /home/kabir/PX4/Firmware/cmake
DF Drivers: ms5607;mpu6050;ak8963;bebop_bus;bebop_rangefinder;mt9v117
-- The ASM compiler identification is GNU
-- Found assembler: /home/kabir/PX4/toolchains/gcc-linaro-arm-linux-gnueabihf/bin/arm-linux-gnueabihf-gcc
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- C compiler: arm-linux-gnueabihf-gcc (Linaro GCC 5.3-2016.05) 5.3.1 20160412
-- C++ compiler: arm-linux-gnueabihf-g++ (Linaro GCC 5.3-2016.05) 5.3.1 20160412
```

RPi builds with new toolchain : 

```
-- PX4 VERSION: v1.6.0rc1-104-g0490bfe
-- CONFIG: posix-rpi-cross
-- CMAKE_INSTALL_PREFIX: /usr
-- CMAKE_MODULE_PATH: /home/kabir/PX4/Firmware/cmake
DF Drivers: mpu9250;lsm9ds1;ms5611;hmc5883;trone;isl29501
-- The ASM compiler identification is GNU
-- Found assembler: /home/kabir/PX4/toolchains/gcc-linaro-arm-linux-gnueabihf/bin/arm-linux-gnueabihf-gcc
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- C compiler: arm-linux-gnueabihf-gcc (Linaro GCC 5.3-2016.05) 5.3.1 20160412
-- C++ compiler: arm-linux-gnueabihf-g++ (Linaro GCC 5.3-2016.05) 5.3.1 20160412

```

mhkabir, 2017-02-05:
Work in progress here: https://github.com/PX4/Firmware/compare/master...mhkabir:skycontroller

Toolchain upgrade complete and tested on all platforms, working on complete Skycontroller solution now.

nicolaerosia, 2017-02-05:
The latest Linaro ARM GCC is 6.2.1: https://releases.linaro.org/components/toolchain/binaries/latest/arm-linux-gnueabihf/

nicolaerosia, 2017-02-05:
@mhkabir might be worth tuning the compilation flags specifically for that ARM platform.
Could you please share `cat /etc/cpuinfo` ?

mhkabir, 2017-02-05:
@nicolaerosia I'm aware that 6.2.1 is the latest. PX4 tends to lag by 1 major release in order to avoid bugs in the bleeding-edge. It would also be quite some effort to update stuff and get everything compiling on 6.2.1. e.g NuttX based build uses 5.4.

Therefore, for now I propose to use 5.3.1. It would be trivial to update to 6.2.1 later.

I agree that platform specific optimization flags make sense. Will share `cpuinfo` in a bit, stay tuned.

nicolaerosia, 2017-02-05:
@mhkabir I did not know that, thanks for letting me know!
Could you share the cpuinfo on both the drone and Skycontroller? Thanks!

mhkabir, 2017-02-05:
@nicolaerosia 

Bebop and Skycontroller use the same SoC.

From Bebop :
```
Processor	: ARMv7 Processor rev 7 (v7l)
processor	: 0
BogoMIPS	: 1561.67

processor	: 1
BogoMIPS	: 1560.02

Features	: swp half thumb fastmult vfp edsp thumbee neon vfpv3 tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x2
CPU part	: 0xc09
CPU revision	: 7

Hardware	: Milos board
Revision	: 0003
Serial		: 0000000000000000

```

From Skycontroller : 
```
root@mppboard:/ # cat /proc/cpuinfo
Processor	: ARMv7 Processor rev 7 (v7l)
processor	: 0
BogoMIPS	: 1562.02

processor	: 1
BogoMIPS	: 1560.01

Features	: swp half thumb fastmult vfp edsp thumbee neon vfpv3 tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x2
CPU part	: 0xc09
CPU revision	: 7

Hardware	: Mpp board
Revision	: 0007
Serial		: 0000000000000000

```

mhkabir, 2017-02-05:
Raspberry Pi 3 : 

```
processor   : 0
model name   : ARMv7 Processor rev 4 (v7l)
BogoMIPS   : 38.40
Features   : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer   : 0x41
CPU architecture: 7
CPU variant   : 0x0
CPU part   : 0xd03
CPU revision   : 4

processor   : 1
model name   : ARMv7 Processor rev 4 (v7l)
BogoMIPS   : 38.40
Features   : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer   : 0x41
CPU architecture: 7
CPU variant   : 0x0
CPU part   : 0xd03
CPU revision   : 4

processor   : 2
model name   : ARMv7 Processor rev 4 (v7l)
BogoMIPS   : 38.40
Features   : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer   : 0x41
CPU architecture: 7
CPU variant   : 0x0
CPU part   : 0xd03
CPU revision   : 4

processor   : 3
model name   : ARMv7 Processor rev 4 (v7l)
BogoMIPS   : 38.40
Features   : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer   : 0x41
CPU architecture: 7
CPU variant   : 0x0
CPU part   : 0xd03
CPU revision   : 4

Hardware   : BCM2709
Revision   : a02082
Serial      : 00000000xxxxxxxx

```

Raspberry Pi 2 : 
```
processor       : 0
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 1
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 2
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 3
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

Hardware        : BCM2709
Revision        : 1a01040
```

nicolaerosia, 2017-02-05:
Thank you! Could you also share dmesg? I'm curious to see what SoC they are using.

nicolaerosia, 2017-02-08:
@mhkabir ping

mhkabir, 2017-02-11:
@nicolaerosia https://gist.github.com/mhkabir/29fffaeeadc7765f2c7ba8c0fe70964b

Now I hope that you will make a pull request against this branch with the suggested optimization flags. 8)

stuee123, 2017-11-11:
Hi, is this so i can setup my rpi to control me beebop? i really dont want to spend $300+ on a skycontroller if i can just setup my rpi to do it instead :)

closed 2018-02-21 by PX4BuildBot