PX4/PX4-Autopilot #6516: Error on px4io-v2/board_config.h ?

matanhavi, 2017-02-04:
I am not sure, but think that there some error on  px4io-v2/board_config.h
The line: 
#define GPIO_PPM_IN		(GPIO_ALT|GPIO_CNF_INPULLUP|GPIO_PORTE|GPIO_PIN9)
says the IO chip has PORT E, but the chip don't have such port.

Its seems that there are two location on the code that refer to this macro:
1) fmu.cpp, (correct if I am wrong) the io code doesn't use it
2) drv_hrt - that seems to configure wrong pin. but because GPIO_PPM is also defined, and initialized there is no issue. 

I might be wrong, and its some kind of trick, and I am not sure if and how it effect the current code, and currently don't have PPM remote to check that.

davids5, 2017-03-21:
@matanhavi 

Thank you for reporting this.

mhkabir, 2017-04-03:
Fixed! Thanks @matanhavi.

closed 2017-04-03 by mhkabir