PX4/PX4-Autopilot #6517: v1.6.0 release stack crash on pixhawk

jgoppert, 2017-02-05:
I'm currently bisecting. 1.5.5 worked still.

jgoppert, 2017-02-05:
First bad commit:

```bash
07923a86c7d83bb2c91bcc2a529b67445e52b97c is the first bad commit
commit 07923a86c7d83bb2c91bcc2a529b67445e52b97c
Author: David Sidrane <david_s5@nscdg.com>
Date:   Mon Dec 12 13:19:49 2016 -1000

    Upgrade to Nuttx 7.18+ ==upstream

:100644 100644 513501ab5c914bc5ed372c2d59f7be1780980665 97df0ea259cc4e7e808da93fb0b7dd70a36cd42f M	.gitmodules
:160000 160000 af5d2b058c55431607aa5aa29fc1d1bbc7e51089 44ad7e224c1ef17911ab8b4101fd624ad9ee4177 M	NuttX
```

@davids5 can you take a look?

dagar, 2017-02-05:
Both current master and the v1.6.0rc1 tag are working on my pixhawk.

Do you want to try one of the build system binaries to see if it's something wrong in your dev environment?

Did you cleanup submodules between bisect tests? `make distclean`

Do you get any stack trace or anything to work with?

jgoppert, 2017-02-05:
Thanks for testing it. I will try one of the binaries, what gcc are you using?

```bash
gcc --version
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.4) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

jgoppert, 2017-02-05:
I'm downloaded the other compiler now: sudo add-apt-repository --remove ppa:team-gcc-arm-embedded/ppa

jgoppert, 2017-02-05:
@dagar the generated firmware in the .zip for the release also fails, this is an official 3dr pixhawk, and it will run 1.5.5, am I missing something? @LorenzMeier

jgoppert, 2017-02-05:
Here is the version info from nsh for the board when running the 1.5.5 release

```
nsh> ver all
HW arch: PX4FMU_V2
FW git-hash: 00334ad76d03b5abc2144f1ebba7faf691448f5c
FW version: v1.5.4-1-g00334ad (1.5.4 0), 17105920
OS version: 6.27 (100796415)
Build datetime: Feb  5 2017 04:14:12
Toolchain: 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]
MCU: STM32F42x, rev. 1

WARNING   WARNING   WARNING!
Revision 1 has a silicon errata
This device can only utilize a maximum of 1MB flash safely!
https://pixhawk.org/help/errata

UID: 310041:3034510C:36353832 
nsh> 
```

jgoppert, 2017-02-05:
It works on another board. It must be hardware issue on my end.
```
nsh> ver all
HW arch: PX4FMU_V2
FW git-hash: 602279ad56d94d8c9943b6ababeb9691160a5905
FW version: 1.6.0 0 (17170432)
OS: NuttX
OS version: Release 1.8.0 (17301759)
Build datetime: Feb  5 2017 04:28:12
Build uri: localhost
Toolchain: GNU GCC, 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]
MFGUID: 363538323034510600300021
MCU: STM32F42x, rev. 1

WARNING   WARNING   WARNING!
Revision 1 has a silicon errata:
This device can only utilize a maximum of 1MB flash safely!
https://pixhawk.org/help/errata

UID: 300021:30345106:36353832 
```

closed 2017-02-05 by jgoppert

davids5, 2017-02-05:
@jgoppert  

If you loan me your "bad" board I can see what is happening.  I will have to add Jtag headers if it does not have them, but I can send it back once it is understood. snailmail address in skype.

jgoppert, 2017-02-05:
Thanks @davids5 I circled back and now it works and I can't reproduce on the same board. I'm going to chalk it up to a late night, still a complete mystery to me, had to be something stale like params on the board maybe.