PX4/PX4-Autopilot #6518: Inspect takeoff handling

LorenzMeier, 2017-02-05:
This PR: https://github.com/PX4/Firmware/pull/6505

Might have introduced a regression. It seems like the handling is not what the operator expects: "first waypoint is takeoff, copter goes to takeoff point rather than first waypoint".

Not yet entirely sure how this change resulted in this, but interesting to note.

http://review.px4.io/plot_app?log=407e9c26-bbc1-4d30-a6b8-c3b528241b43

LorenzMeier, 2017-02-05:
@Santiago3DR I would expect the system to go to the "T" symbol on the map in QGC. Is this what you expect? Or do you expect it to go to go to the first waypoint? I think we do not expect it to skip the takeoff waypoint ever, but to always go there.

santiago3dr, 2017-02-05:
Even if already in air?

LorenzMeier, 2017-02-05:
Yes. The goal is to have certainty that the drone passes through that point. Because you might have planned a mission that first needs to climb to 30 meters before doing the survey. If you skip to the survey / later waypoints directly you will still be climbing while moving forward, potentially hitting obstacles.

It sounds like we can close this.

closed 2017-02-05 by LorenzMeier