PX4/PX4-Autopilot #6519: uavcan fw server does not start automatically when pixhawk reboots

tstellanova, 2017-02-05:
I have UAVCAN_ENABLE set to 3
 I can see that the `uavcan` task is running (using `top`); however `uavcan fw srv` is not running.

IF I subsequently `uavcan start fw` the server starts up OK.  

Tested with both Pixhawk 2.4 and Pixracer R14 hardware, latest PX4/Firmware origin/stable

LorenzMeier, 2017-02-05:
Are you on a VTOL? What is the output of "free"?

tstellanova, 2017-02-05:
Not on VTOL, just a straight fresh build of origin/stable (v2 or v4, your choice)
```
nsh> free
             total       used       free    largest
Mem:        231328     200544      30784      15056
```

pavel-kirienko, 2017-02-05:
The output of `free` looks sane, we're checked it over at https://gitter.im/UAVCAN/general.

pavel-kirienko, 2017-02-05:
Todd, UAVCAN has been dropped from v2, you're supposed to use v3.

tstellanova, 2017-02-05:
Ok, I'm happier to focus on v4 since that's easier to debug

LorenzMeier, 2017-02-05:
Yep, that's the culprit on v2. On v4: Ensure that `uavcan start` does return 0, not any error code:

```
#
	# Starting stuff according to UAVCAN_ENABLE value
	#
	if param greater UAVCAN_ENABLE 0
	then
		if uavcan start
		then
			uavcan start fw
		else
			tone_alarm ${TUNE_ERR}
		fi
	fi
```

tstellanova, 2017-02-05:
FWIW If I stop and then restart uavcan from nsh, there don't appear to be any errors.

mhkabir, 2017-02-05:
Have a bootlog from the console?

tstellanova, 2017-02-05:
I'll collect a console bootlog within 24 hrs.
I don't recall seeing any uavcan errors scroll by...

tstellanova, 2017-02-06:
Bootlog (pixracer/v4):
```
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
WARN  [param] selected parameter default file /fs/mtd_params
WARN  [rgbled] no RGB led on bus #1
WARN  [blinkm] I2C init failed
WARN  [blinkm] init failed
INFO  [ver] match: PX4FMU_V4
INFO  [ver] match: PX4FMU_V4
INFO  [dataman] Unkown restart, data manager file '/fs/microsd/dataman' size is 47640 bytes
INFO  [ver] match: PX4FMU_V4
WARN  [drivers__boards__px4fmu-v4] reset done, 50 ms
ERROR [ms5611] driver start failed
MS5611_SPI on SPI bus 2 at 3 (20000 KHz)
WARN  [bst] no devices found
INFO  [ver] match: PX4FMU_V4
WARN  [hmc5883] no device on bus 2
read_reg failWARN  [lis3mdl] no device on bus 2
LIS3MDL_SPI on SPI bus 1 at 7 (11000 KHz)
LIS3MDL bad ID: ffWARN  [lis3mdl] no device on bus 3
HMC5883_SPI on SPI bus 1 at 5 (11000 KHz)
MPU6000 on SPI bus 1 at 6 (1000 KHz)
MPU9250 on SPI bus 1 at 4 (1000 KHz)
ERROR [bmi160] driver start failed
ERROR [meas_airspeed] no MS4525 airspeed sensor connected
ERROR [ets_airspeed] no ETS airspeed sensor connected
ERROR [ets_airspeed] no ETS airspeed sensor connected
SF1XX disabled.
ERROR [sf1xx] driver start failed
WARN  [hmc5883] mag cal status changed 
WARN  [commander] No RGB LED found at /dev/rgbled0
nsh: px4io: command not found
INFO  [ver] match: PX4FMU_V4
INFO  [mavlink] mode: Config, data rate: 20000 B/s on /dev/ttyS0 @ 921600B
WARN  [mavlink] hardware flow control not supported
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
INFO  [mavlink] mode: OSD, data rate: 1000 B/s on /dev/ttyS2 @ 57600B
WARN  [uavcan] Node ID 2, bitrate 1000000
WARN  [uavcan] CAN driver init...
WARN  [uavcan] SW version vcs_commit: 0x00334ad7
WARN  [uavcan] sensor bridge 'gnss' init ok
WARN  [uavcan] sensor bridge 'mag' init ok
WARN  [uavcan] sensor bridge 'baro' init ok
INFO  [ver] match: PX4FMU_V4
INFO  [ver] match: PX4FMU_V4
px4flow [210:100]
WARN  [px4flow] scanning I2C buses for device..
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
WARN  [uavcan] subscribe to actuator_controls_0
WARN  [uavcan] subscribe to actuator_controls_3
INFO  [init] Mixer: /etc/mixers/quad_x.main.mix on /dev/uavcan/esc 
INFO  [lib__ecl] EKF IMU buffer length = 21
INFO  [lib__ecl] EKF observation buffer length = 16

NuttShell (NSH)
nsh>
```

tstellanova, 2017-02-06:
BTW here's what the code in stable branch shows for me:

```
	#
	# Starting stuff according to UAVCAN_ENABLE value
	#
	if param greater UAVCAN_ENABLE 0
	then
		if uavcan start fw
		then
		else
			tone_alarm $TUNE_ERR
		fi
	fi
```

It appears that calling `uavcan start fw` from nsh does not start the firmware (node ID) server if uavcan has not already been started.

tstellanova, 2017-02-06:
Built and installed px4fmu-v4_default from master branch.  This problem is apparently fixed in master.  Not sure how this project handles rolling bug fixes into the stable branch these days, but this seems like a good candidate for a patch.

mhkabir, 2017-04-03:
This is fixed.

closed 2017-04-03 by mhkabir