PX4/PX4-Autopilot #6525: In the Qualcomm_Snapdragon_Flight, ESC can't working

sunshine-yan, 2017-02-06:
Hi,
First of all, i have used this code in the  Qualcomm_Snapdragon_Flight.
Now, i have tried to use two sets of code. One of them is this PX4/Firmware. However, the ESC can't working  when i use this code. I find all of the code, I think 'snapdragm_pwm_out' may be related to ESC.Therefore I have tried to replace the "snapdragm_pwm_out start -d /dev/tty-2" with "uart_esc start -D /dev/tty-2". However,ESC still can't start.
The situation is as follows：
[image]

Stifael, 2017-02-07:
To what ports of the snapdragon did you attach the esc board?

sunshine-yan, 2017-02-07:
@Stifael It's this port.As shown in the yellow box.
[image]

Stifael, 2017-02-07:
This is correct. Does it work now? Or why did u close the issue?

sunshine-yan, 2017-02-07:
@Stifael I'm sorry, I just accidentally closed the issue.It still can't work.

Stifael, 2017-02-07:
So if you want to use snappy to send out pwm signals, then you should use snapdragon_pwm_out. I need to retest it my own to see if actually nothing broke, but I suggest to look at the pin outputs to see if there is any signal.

sunshine-yan, 2017-02-07:
@Stifael I I have tried to use snapdragon_pwm_out.However,the ESC did not respond.LEDs in the ESC don't flash green. One LED flash red.So i don't know how to judge there is no signal.

Stifael, 2017-02-07:
@sunshine-yan: I actually meant to look at the signal at the snapdragon port. that is independent of the esc. If the snapdragon does not send any pwm singals, then obviously the esc is not responding either. 

Before I test, I would just be sure that your are also tested with the newest master. Is that correct?

sunshine-yan, 2017-02-07:
@Stifael How do I detect if  snapdragon send any pwm singals? Is there any place to print?

Stifael, 2017-02-07:
I mean you could do PX4_INFO in snapdragon_pwm_out (https://github.com/PX4/Firmware/blob/master/src/drivers/snapdragon_pwm_out/snapdragon_pwm_out.cpp#L317) and see what it prints. But before you do that, did you look at the mini-dm what the output is? does it say that the snapdragon_pwm_out is initialized correctly, or is there any error message?

sunshine-yan, 2017-02-07:
@Stifael Now it is error. Such as the following picture:
[image]

Stifael, 2017-02-07:
I just tested. so the default is set to pwm_out_rc_in. change that to snapdragon_pwm_out as you did, but just write start without defining the port:
snapdragon_pwm_out start

sunshine-yan, 2017-02-08:
@Stifael Thanks for your help. I try to write: snapdragon_pwm_out start
[image]
Add output to the code:
[image]
The situation is as follows：
[image]

Seanmatthews, 2017-03-02:
Bump. I've talked to a handful of people with this issue. Any progress?

julianoes, 2017-03-03:
Guys you are talking about two different types of ESCs.

1. PWM support that was recently added (@Stifael I think that's what you're refering to).
   needs `snapdragm_pwm_out start -d /dev/tty-2`
2. Snapdragon ESCs connected over UART.
    needs `uart_esc start -D /dev/tty-2`

@sunshine-yan what is the output in mini-dm when you try to start the uart_esc like this:

```
qshell uart_esc start -D /dev/tty-2
```

mhkabir, 2017-04-03:
Closing in favour of #6962

alexandrelheinen, 2017-05-02:
I am having a similar issue:

I am trying to implement a PX4-based autopilot for a fixed-wing UAV. It turns out that the easiest alternative is to use the 4 PWMs output on port J13 which uses `snapdragon_pwm_out` module (https://github.com/PX4/Firmware/tree/master/src/drivers/snapdragon_pwm_out).

As we can observe in the source code, it tries to parse the mixer file into a `MultirotorMixer` object. In practice, I am having the following error from the mini-dm
```
[08500/00]  01:50.030  HAP:34:Trying to initialize mixer from config file /dev/fs/myfixedwing.main.mix  0166  snapdragon_pwm_out.cpp
[08500/03]  01:50.032  HAP:34:Unable to parse from mixer config file  0181  snapdragon_pwm_out.cpp
[08500/03]  01:50.032  HAP:34:Mixer initialization failed.  0337  snapdragon_pwm_out.cpp
```
even if I use the example mixer file given in the DevGuide (https://dev.px4.io/airframes-adding-a-new-frame.html).

Does it have any link with the class `MultirotorMixer`? Could I initialize the fixed-wing mixer without using a `.mix` file, setting up the `MultirotorMixer` directly?

LorenzMeier, 2017-05-02:
Yeah that's clearly wrong try replacing this here:
https://github.com/PX4/Firmware/blob/master/src/drivers/snapdragon_pwm_out/snapdragon_pwm_out.cpp#L174

With the more generic call Mixer::from text or something along these lines:
https://github.com/PX4/Firmware/blob/master/src/drivers/navio_sysfs_pwm_out/navio_sysfs_pwm_out.cpp#L135-L164

alexandrelheinen, 2017-05-03:
UPDATE: the reason why I am getting this error is why the `MultirotorMixer::from_text` function (https://github.com/PX4/Firmware/blob/master/src/modules/systemlib/mixer/mixer_multirotor.cpp) requires a mixer file defines the Multirotor Geometry in the first line. However how could I do this from a fixed-wing, if only multirotors geometries are accepted?

If a give an unrecognised geometry, what are the consequences on the mixer suitability?

closed 2018-02-21 by PX4BuildBot

julianoes, 2018-02-21:
Snapdragon Flight is end-of-lifed, anyway, not sure if that's worth the effort, so keeping it closed.