PX4/PX4-Autopilot #6533: Mc not holding position during takeoff

sanderux, 2017-02-07:
Since a few weeks master isnt holding position during takeoff
Also it no longer responds to a level command making mission flight impossible and dangerous

@LorenzMeier @dagar @tumbili @AndreasAntener

sanderux, 2017-02-07:
I'm in senegal making it hard to debug, log will follow

sanderux, 2017-02-08:
Log: http://logs.uaventure.com/view/tLhQrFFedHMEQHxCdqNrae#Local_X_PLOT
Note the drift in pos x/y

AndreasAntener, 2017-02-08:
Pretty sure that was because of motor saturation. It didn't do a takeoff but instead navigated to the waypoint normally (high Z velocity setpoint). What do you mean with `level command`?

sanderux, 2017-02-08:
Hm, it did hit the sky pretty quick failing to agnowledge the takeoff speed. I'll see whats going on there. 

Level horizon doesnt respond in qgc on tablet or pc, usb or 3g telem

sanderux, 2017-02-08:
This does expose another debatable issue, apperantly altitude takes precedence over horizontal position.

AndreasAntener, 2017-02-08:
Yes this is an issue and already a long standing discussion. Let me see if/where we track this already.

Can you open an issue on the level horizon? I've heard that from someone on gitter too.

closed 2017-02-08 by AndreasAntener