PX4/PX4-Autopilot #6534: Fixed Wing HiL controls broken in 1.6.0rc1

aivian, 2017-02-07:
On 1.6.0rc1 the actuator controls are all zeros coming out of the pixhawk. Controls are zeroed when commands originate from the pixhawk (in stabilized/hold mode) and under manual control (using a receiver on futaba SBUS).

Versions up to 1.5.5 work well.

A diff on the two branches indicates that a bunch of UART rc input stuff was added to simulator_mavlink.cpp, so that was my first thought. However, removing all code inside of a ENABLE_UART_RC_INPUT ifdef block did not fix the problem.

aivian, 2017-02-07:
Nevermind, turns out this is another manifestation of these two

https://github.com/PX4/Firmware/issues/5680
https://github.com/PX4/Firmware/issues/6374

Closing issue in deference to the already open ones.

closed 2017-02-07 by aivian

goodman110110110, 2017-03-03:
which version of the QGC?

aivian, 2017-03-06:
The problem is in the px4 firmware and is unrelated to qgc. This should be fixed in the current master.

goodman110110110, 2017-03-14:
what is the reason of the problem?
Pixhawk firmware will not support the flightgear Hil? The current master
has not the flightgear hil airframe but xplane. How could I do to do the
hil with flightgear?

aivian, 2017-03-15:
I'm not sure what you mean. The px4 firmware doesn't have specific configurations for different HiL simulations. This issue was around a flag that was not being set properly for HiL simulation of fixed-wing aircraft. This issue was present for all simulation environments, and has been fixed.

Separately, and likely the source of your issue is that the interface between qgc and flightgear is broken right now after the change to actuator_controls. Most people doing hil are using xplane so no one has submitted an actuator_controls patch for the flightgear interface yet. That isn't relevant to the px4 firmware or this issue however.

goodman110110110, 2017-06-11:
In the firmware files which is init.d, there is the the
file"1004_rc_fw_Rascal110.hil", so px4 has not support the flightgear
simulation.