PX4/PX4-Autopilot #6541: Bug in ms5611 driver framework

langxm1223, 2017-02-08:
I tested the ms5611 on snapdragon board with official PX4 firmware.
The attached is the log from the df_ms5611_wrapper.cpp
[image]

As you can see there are many error values in the log.
I think there must be some bug in the driver.
Hope someone could solve it.
Thanks.

LorenzMeier, 2017-02-08:
Did you try latest master (from today)? We fixed an issue recently thanks to @dxwang.

LorenzMeier, 2017-02-08:
It was a timing fix. Please retry and see if its still the same.

langxm1223, 2017-02-09:
Hello @LorenzMeier 
I tried the latest DriverFramework submodule.
The same problem can be observed as well.
Actually, there is no update related to ms5611 driver from the commit I used to the latest commit.
Any ideas?
Thank you very much.

LorenzMeier, 2017-02-17:
There was a fix and we validated it to be good. Can you share the commit you tested? And would you mind updating all submodules and trying master again with a clean build?

langxm1223, 2017-02-20:
Hello @LorenzMeier 
I checked and tested it again.
The commit of driver framework I used is 0b9afafa2e05ae3899c6d38231196ace265003b2
The bug still existed.
Hope that someone could take a more thorough test.
Thank you very much.

dagar, 2017-04-11:
@langxm1223 any update? Have you tried this recently?

langxm1223, 2017-04-14:
@dagar Sorry for the delay.
Still the same here.
Furthermore, I also observed some nan value of the pressure variable from the sensor driver.
No clue yet, probably bug in memory copy.
Thanks.

closed 2018-02-21 by PX4BuildBot