PX4/PX4-Autopilot #6545: Intel Aero Barometer offset

ChristophTobler, 2017-02-09:
This issues replaces https://github.com/intel-aero/meta-intel-aero/issues/18

lucasdemarchi, 2017-02-09:
#6548  fixes the huge offset you see.

ChristophTobler, 2017-02-09:
I can confirm that https://github.com/PX4/Firmware/pull/6548#pullrequestreview-20959539 fixes the offset. The noise remains though

davids5, 2017-02-09:
The noise is due that fact the MS56xx is on the same SPI bus as the the other sensors.
[image]

The baro should be moved to from SPI3 to SPI2 or better yet it's own spi bus.

lucasdemarchi, 2017-02-09:
Baro and mag are physically on the same **I2C** bus, with no way to move it
elsewhere.

A patch was already merged to disable the mag, so right now we are only
talking to the baro.

davids5, 2017-02-09:
Sorry I look at the there aero  board config.

LorenzMeier, 2017-04-12:
Fixed.

closed 2017-04-12 by LorenzMeier