PX4/PX4-Autopilot #6547: Intel Aero IMU noise

ChristophTobler, 2017-02-09:
@lucasdemarchi This issues tracks https://github.com/intel-aero/meta-intel-aero/issues/20

lucasdemarchi, 2017-02-09:
#6548  fixes the huge offset you see. Noise should not be affected though

mhkabir, 2017-04-26:
[image]

mindThomas, 2017-04-26:
To pitch in here, I would strongly suggest that you find a better method to fasten the Intel Aero Compute board with the flight controller within the plastic enclosure on the RTF drone.
Currently the Aero compute board is just laying loosely in the plastic frame, only fixed by the four screw holes. This does only fix the horizontal alignment more or less though it does not fix the z-direction in any way why accelerations of the frame would easily propagate and result in the Intel Aero board vibrating.

We will try with some sticky tape/foam to hold the board down inside the plastic enclosure and see if this improves anything.

closed 2018-02-21 by PX4BuildBot