PX4/PX4-Autopilot #6549: Not responding to level horizon command

sanderux, 2017-02-09:
Current master (20160208) does not respond to level horizon command from QGC
Only verified on VTOL vehicle
All other sensor calibrations work correctly
QGC does not show a progress bar and no info messages on the matter are being sent making it appear as if the command was never received. However, the QGC version used to test has successfully ran this command on earlier versions

/cc @AndreasAntener

sanderux, 2017-02-14:
@DonLakeFlyer any idea?

DonLakeFlyer, 2017-02-14:
Debugged through QGC side. Calibration message is being sent by QGC but no response from firmware.

sanderux, 2017-02-14:
@tumbili can this be related to your recent commit?
https://github.com/PX4/Firmware/commit/afb01e6d9aab2e7feb32125948d55d0066565a9e

sanderux, 2017-02-14:
@bugobliterator bisecting this issue points to any of your following commits:

https://github.com/PX4/Firmware/commit/c9ac15f0ddb54483194e4c8b14e3f246338c9f5a
https://github.com/PX4/Firmware/commit/f811777789ed9901e7f94d3a1d9f4adfbdd8beb4
https://github.com/PX4/Firmware/commit/78b8deda1592f16785d585d456bc2d294767f253

Can you review?

bugobliterator, 2017-02-15:
@sanderux sure I will have a look

LorenzMeier, 2017-02-15:
Fixed in https://github.com/PX4/Firmware/commit/081c40c6fa2045560f80b0c0707f400eb31b7ff4

closed 2017-02-15 by LorenzMeier