PX4/PX4-Autopilot #6554: VTOL Massive altitude increase after backtransition

sanderux, 2017-02-09:
http://logs.uaventure.com/view/Hw5VgbagMaWvCWxMPVLAsA#Local_Z_PLOT
Perhaps some regression.

sanderux, 2017-02-09:
@dagar any idea?

dagar, 2017-02-09:
FW_PSP_OFF 0?

dagar, 2017-02-09:
Nevermind, I guess that's your typical setup.

What's going on with the thrust setpoint?

[image]

dagar, 2017-02-09:
@Stifael would you mind taking a quick look at this log from the multicopter position controller perspective? 
After back transition finishes (16:01:17) it's effectively a regular multicopter in AUTO.

[image]

sanderux, 2017-02-10:
@dagar not my typical setup but i cant change the horizon level at the moment :)

sanderux, 2017-02-10:
@dagar could this be related? https://github.com/PX4/Firmware/commit/480dd0922b65a00a8e5424d475b5b32c57b27bc2#diff-bcb38f37ec5bebecc13965e3fec8c312R292

sanderux, 2017-02-10:
Or this: https://github.com/PX4/Firmware/commit/f297c45f78e4f487da2c2485588ef061eef1fcd2

sanderux, 2017-02-10:
@MaEtUgR could your recent landing changes be responsible for this behavior?

Stifael, 2017-02-13:
https://github.com/PX4/Firmware/commit/f297c45f78e4f487da2c2485588ef061eef1fcd2 should only be entered if velocity in z is below some threshold and the user throttle stick is put to min. In addition, the part of the code should only affect the thrust setpoint in x and y. 
I need to have a closer look at the transition to understand why the thrust is set to such a large value after transition

sanderux, 2017-02-13:
@AndreasAntener could this logic https://github.com/PX4/Firmware/commit/5f3cbbbbc2c9065a54443fa6cc91cc3b16ef4d89#diff-fc77c3ef569029d45764664c75d4b0c1R1911 break down in weathervane mode and cause this behavior?

dagar, 2017-02-14:
@sanderux have you tried to reproduce in SITL?

sanderux, 2017-02-14:
Yes, but havent managed to reproduce yet

Stifael, 2017-02-14:
5f3cbbb#diff-fc77c3ef569029d45764664c75d4b0c1R1911 is just a projection and should even reduce the thrust for large angle error (assuming there is no logical error in there). What seems strange to me is that you were not able to reproduce it in sitl. I mean it is clear that after transition there is no tracking in the z direction due to the large thrust, which might have accumulated to the max due to the integration.

sanderux, 2017-02-14:
Yes, i have just attempted to run SITL again with the sens_board_y_off at 5deg to simulate the circumstances it had (level horizon no longer works and the plane was leveled for optimal FW flight, for MC flight would fly level when pitch was 5deg down)

Still does not reproduce this. i am flying in SITL the same branch that was used for this flight.

sanderux, 2017-02-14:
Another point to note is that SITL reduces X/Y velocity much faster in back transition compared to a real VTOL vehicle. The full thrust setpoint seemed to be active while the x/y velocities were too high. as soon as they started tracking the thrust lowered:

[image]

Stifael, 2017-02-14:
ok, I also just tried. the  issue that I encountered was at the end of the flight the vehicle was not able to hold its position, but I think that is unrelated to this issue since after transition the thrust setpoint was within reasonable bounds. 
from your plots: that is exactly why we also need to have velocity setpoint and velocity on the same graph (which currently is not the case for flightreview nor logmuncher).
But how was is in the past? I assume this large velocity error was always the case. and 5f3cbbb#diff-fc77c3ef569029d45764664c75d4b0c1R1911 should even decrease the thrust setpoint and favor attitude alignment. 
I guess one problem is that there is only a slewrate change limit for velocity setpoint, but no acceleration limit. the slewrate ensures that the velocity setpoint only changes with some defined max, but it changes feedforward-like without considering the actual vehicle state. this can be seen from your plot, LPSP.VY at ~730, where the setpoint changes with some max slope. However, if the controller is not able to keep up as it is in ~730, then the controller t asks for very large acceleration.

sanderux, 2017-02-14:
here's a log for a similar flight on a jan 12th master: http://logs.uaventure.com/view/6cqCEALzauAUoVzRDnPxkP

There it had no such behavior and the velocities were also too high for approximately the same duration:
[image]

Stifael, 2017-02-14:
well the velocity tracking seems much better. I mean the max velocity error in y for the second plot is around 4, where the max for velocity error for the first plot is more than double (~11)

sanderux, 2017-02-14:
True, it transitioned in a stronger headwind so velocities were lower to begin with. But your idea about velocity setpoint doesnt seem to be the cause as the thrust setpoint was already at max before this became active:

[image]

Stifael, 2017-02-14:
hm, thats also true

sanderux, 2017-02-14:
Also interesting that the thrust setpoint shot to max +- 1 sec after back transition started, and not directly. All this time the altitude was on or above the setpoint

Stifael, 2017-02-14:
Does that not mean that it is some transition issue? Is the mc_pos_control even doing anything during transition.  (https://github.com/PX4/Firmware/blob/master/src/modules/vtol_att_control/vtol_att_control_main.cpp#L777)

sanderux, 2017-02-14:
The thrust setpoint is still published. As you can see the thrust is gradually increased by the vtol controller but it doesnt change the setpoint

sanderux, 2017-02-14:
@Stifael nothing in your commit here that could cause this?
https://github.com/PX4/Firmware/commit/57b7c4fc3db364f014f4bddef0c89eb8f6586c02

Stifael, 2017-02-14:
The comparison between the two commits is actually just a rewrite of something that I changed earlier and should be equivalent. If it used to work in the past, then it makes more sense to compare this change:
https://github.com/PX4/Firmware/commit/f718b3a97ac95ebd8d94d06a85cc3d8180c95476
However, I dont think that this change has any effect, because it just limits the change in thrust setpoint.

To be quite honest, I still think that it might be related to the integration. In the past the integration started from zero with no feedforward, now it uses a feedforward thrust of 0.5. This also means that when integration starts and hover thrust is actually below 0.5, the vehicle first will gain in altitude. I really need to look at what happens during transition

dagar, 2017-02-14:
@Stifael the main thing that happens during FW -> MC transition is that position and velocity control are disabled. https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L3687
The multicopter position controller is largely oblivious to VTOL, other than resetting the yaw and alt setpoints during FW mode.

MC actuators are blended back on proportional to airspeed or timeout (no airspeed sensor). https://github.com/PX4/Firmware/blob/master/src/modules/vtol_att_control/standard.cpp#L421

dagar, 2017-02-14:
Regardless of this bug, shouldn't the integrators be reset?

Either make this section apply to VTOL FW mode - https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L2279

or reset here - https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L2256

Stifael, 2017-02-14:
Ok, if pos and vel control are disabled, then it will not enter the above commits anyway. @dagar I agree that it would make sense to reset it also when switching from fw-»mc, if that hasnt be done yet. 
in addition, if something within the mc controller caused this bug, then it has to be related to the attitude - thrust mapping.

sanderux, 2017-02-14:
As for the position control during transition @dagar is correct, we should be looking at attitude control for where it starts, but the problem remains long after the transition so that would point to integrator issues. And considering the code dynamics of VTOL integrator issues are quite likely.

The main problem to find this is that this behavior was introduced somewhere between jan-12th and feb-7th and i don't see much more then the commits i already pointed to

sanderux, 2017-02-14:
@dagar although SITL doesnt show to be affected it does exhibit some strange behavior in thrust setpoint during back transition:

[image]

sanderux, 2017-02-15:
Unable to reproduce, did 2 flights on the same software / coditions
http://logs.uaventure.com/view/K9a83KbPUPzDhC6eDGjQ8P
http://logs.uaventure.com/view/zfssNUUJpYgpt6wNL9tc8Y

sanderux, 2017-02-15:
The increase in altitude in these logs is most likely related to the fact that level is off but the thrust setpoint issue did not occur

dagar, 2017-02-15:
I think we should thoroughly reset the internal flags in the multicopter position controller during VTOL FW mode, possibly add additional logging, and move on.
https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L2305

@Stifael is there any additional logging that you would have found useful for debugging this?

sanderux, 2017-02-15:
Doesnt this https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L3683
and this: https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L2279

make the controller not reset?

dagar, 2017-02-15:
@sanderux I just glanced at your logs and the altitude during back transition is terrible. 

[image]

Why does the MC thrust remain high as you continue to overshoot the altitude?

[image]

We should also modify fw_att_control to not reset the integrators until back transition has finished. That's why you pitched up suddenly. Look at the jump in ATC1.Pitch even though the pitch setpoint didn't change.

dagar, 2017-02-15:
> Doesnt this https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L3683
and this: https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L2279
make the controller not reset?

I'm talking about resetting like the section below.

closed 2017-02-28 by Stifael