PX4/PX4-Autopilot #6555: LOCAL_POSITION_NED.z not stable, varying erratically.

nahush8, 2017-02-09:
We are trying to do off-board control for Pixhawk using MOCAP. Although we are getting fairly good estimates for x and y positions, z position varies abruptly. Attaching the 'z' plot from QGC for the vehicle at rest.  
Even after incorporating fake gps, and setting EKF2_GPS_V_NOISE to low, the problem persists.  

@mhkabir, I have read similar issues before and couldn't really find any solutions for the same. Could you shed some light on this ? 

Firmware version: 1.5.5 
QGC version: v3.1.2
[image]

jgoppert, 2017-02-09:
Most people using mocap have been using LPE. You might want to try that instead. It is in the latest release zip file and also you can build/upload it with: make px4fmu-v2_lpe upload

nahush8, 2017-02-10:
@jgoppert : So we've tried using LPE and we observed similar problems in past. Is this zip the latest firmware ? 1.6.0 ?

jgoppert, 2017-02-10:
@nahush8 Please post a full log, then maybe we can help figure out why. I know some others have had this working.

mhkabir, 2017-02-10:
Z issues are because its trusting the baro. @jgoppert Can you add the LPE parameters that need to be changed to get it to not fuse the baro (the bitmask) and trust vision/mocap more (std. devs)? This page : https://dev.px4.io/external-position.html

nahush8, 2017-02-10:
@mhkabir  So the new firmware does not seem to show ATT_EXT_HDG_M in Qgroundcontrol, so we couldn't find any way of setting it to mocap. We tried making the EKF Baro noise high, that doesn't seem to work either.

mhkabir, 2017-02-10:
That's because you're running LPE / Attitude Estimator Q. Do as James says and use the LPE target.

nahush8, 2017-02-10:
okay will try and get back with the results . Thanks !

jgoppert, 2017-02-10:
To disable baro fusion it is now just a check box  for  param LPE_FUSION.

nahush8, 2017-02-10:
@mhkabir  So I uploaded the given firmware for lpe and i could find the appropriate parameters now. Also set the LPE_FUSION parameters to not consider Baro. We are facing the "mocap_timeout" issue. 

We had encountered a problem of "mocap_timeout" before while trying this, I did read through some issues before but haven't been able to find a solution for it.

mhkabir, 2017-02-10:
If you're having timeouts, fix your system design. It's because of one these reasons -
1. You don't have enough bandwidth
2. Link latency is too high.
3. Timesync (automatically done if using ROS/Mavros) can't work because of limited bandwidth.

nahush8, 2017-02-10:
@mhkabir  So I tried switching from US connection to telemetry connection (odroid-Pixhawk) and mocap timeout still persists. Could you be able to point me to some post/resources for the bandwidth issue you mentioned ? Is there a way to disable a few mavlink topics and would that provide more bandwidth for the position estimate data ?

mhkabir, 2017-02-11:
It's more likely that your latency is too high. Try printing `(hrt_absolute_time() - mocap.timestamp)/1000.0f` in the Mavlink receiver to see what your deltas are like.

nahush8, 2017-02-23:
@mhkabir  So I uploaded the firmware, also my lpe_fusion parameter is 4 (Only vision data). Mavros/local_position/pose and mocap pose match perfectly. (I checked that in RVIZ and also rostopic echo). I am trying to run Position_control. I start flying the quadrotor manually and switch to position_control. As soon as i switch, it drifts in the XY direction, and sometimes in Z direction. I also recorded bags during the experiment and they look okay. (Both pose match). Is there any other configuration on the quadrotor side that I may be missing ?

mhkabir, 2017-02-23:
Please try it out with these 2 bugfixes : 
1. https://github.com/mavlink/mavros/pull/660
2. https://github.com/PX4/Firmware/pull/6647

nahush8, 2017-02-27:
I am finally able to get consistent mocap readings as I said earlier. I am sure I am missing controller tuning to get position control to work fine. Are there any specifi parameters to look for ? And if i need to post logs, what kind of logs do I need to collect and how? thanks

closed 2017-02-27 by nahush8

wuwushrek, 2017-04-08:
Hi !
I have the same problem as you. I'm doing offboard control using  mocap (VICON). I follow all the instructions there and get Mavros/local_position/pose and mocap pose match perfectly. But I can't manage to do position control since the quad just go to the ceil. Have you found any solution ?
Thanks