PX4/PX4-Autopilot #6557: The hovering of iris + hokuyo(lidar) in gazebo is not stable.

hayoonko, 2017-02-10:
I joined hokuyo lidar in iris.sdf. After running gazebo, I just gave commander takeoff command. But the iris drone does not hover. There is no problem when you separate hokuyo and take off only iris. If you know how to fix it, please let me know. The following code shows how to add hokuyo lidar to iris.sdf. Is there a problem with this part? Hokuyo I think it is due to the weight of the weight was set very small, but it was useless.
<include>
      <uri>model://hokuyo</uri>
      <pose> 0 0 0.07 0 0 0</pose>
	<mass>0.01</mass>
    </include>
    <joint name="hokuyo_joint" type="revolute">
      <child>hokuyo::link</child>
      <parent>base_link</parent>
      <axis>
        <xyz>0 0 1</xyz>
        <limit>
          <upper>0</upper>
          <lower>0</lower>
        </limit>
      </axis>
    </joint>

mhkabir, 2017-02-10:
Tried zero-ing the mass?

hayoonko, 2017-02-10:
@mhkabir  yes, I have tried the mass of hokuyo lidar at zero, but it has not been hovered.

jgoppert, 2017-02-10:
I created a new iris model with an rplidar here: https://github.com/PX4/Firmware/pull/6558

The main issue with your model is you forgot to set the inertia matrix terms, they default to 1.

hayoonko, 2017-02-10:
@jgoppert I just added your sdf and checked it and it is in perfect hovering state. I really appreciate it. As you pointed out the moment of inertia as a problem, can you give me a specific explanation of this part? Thank you once.

jgoppert, 2017-02-10:
Gazebo defaults model inertia matrix to identity if not specified.

BossKwei, 2017-02-13:
I have the same problem:
https://github.com/PX4/sitl_gazebo/issues/88

closed 2017-02-13 by jgoppert

nahush8, 2017-03-08:
@hayoonko  I am trying to attach a lidar to iris and try hovering. I am facing the same issue.  What exactly did you do and which sdf are you referring to ?  Can you tell me some steps of running the entire simulation ? I was following the steps from https://dev.px4.io/simulation-gazebo.html and tried adding rplidar into the models directory of sitl_gazebo

dronecrewkari, 2018-01-30:
@hayoonko  How did you change the inertia matrix?

bresch, 2018-01-30:
@dronecrewkari http://wiki.ros.org/urdf/Tutorials/Adding%20Physical%20and%20Collision%20Properties%20to%20a%20URDF%20Model
As James said, the default inertia matrix if not specified is the identity matrix. Those values are way too big for such a small drone.