PX4/PX4-Autopilot #6559: Fixed wing minimum ground speed

Antiheavy, 2017-02-10:
Feature request:  Add a minimum ground speed setting for fixed wing (new parameter).

Behavior: The fixed wing controller tries to increase airspeed above FW_AIRSPD_TRIM to maintain a minimum ground speed, up to FW_AIRSPD_MAX.

Reasons:  This can be a safety feature for getting home in high winds.  This may help some aircraft make high wind landings more accurate.  This can make mapping missions more effective and efficient by covering ground at a reasonable rate and reducing crab.

Things to consider: Would setting this parameter to 0 effectively disable this feature?  Should 0 be the default?  How would backwards flight affect this?

dagar, 2017-02-10:
The fixed wing controller should already have this (called ground speed undershoot - https://github.com/PX4/Firmware/blob/master/src/modules/fw_pos_control_l1/fw_pos_control_l1_main.cpp#L1038), but it only applies during AUTO.

If you have flight logs from a windy day we should be able to verify if it's actually working as intended.

Antiheavy, 2017-02-10:
We can share some flight logs from today, maybe it is working as intended.  I feel like I'd want to set minimum ground speed explicitly as a parameter.  I'm sort of confused by how "ground_speed_desired" is calculated - there must be some thought behind it here:
https://github.com/PX4/Firmware/blob/master/src/modules/fw_pos_control_l1/fw_pos_control_l1_main.cpp#L1049

dagar, 2017-02-10:
It's mainly this part, with delta_altitude and distance set from either the previous waypoint to your current (if you had a previous), or using your current position to current waypoint.

```c++
float ground_speed_desired = _parameters.airspeed_min * cosf(atan2f(delta_altitude, distance));
```

Following that trajectory at FW_AIRSPEED_MIN would be a groundspeed of ground_speed_desired.

Antiheavy, 2017-02-11:
okay.  Looking back through some flight logs I see this working now.  Actually several things about how TECS is controlling airspeed now make more sense to me.  thanks for the explanation.  looks like this is a non-issue.

closed 2017-02-11 by Antiheavy