PX4/PX4-Autopilot #6565: VTOL Tailsitter instant transition to FW if airpeed is above transition speed

VTOLDavid, 2017-02-11:
I had a crash that would have been prevented if something like this had exist. I lost control flying the tailsitter at high speed in MC mode, 75º pitch, 15m/s, elevons active only for YAW (MC reference system) (maybe my fault for doing this, but under wind conditions being able to fly like that is needed). It ended flying upside down at high speed towards ground and the MC controller unable to control situation, I tried to change to FW flight but nothing happened and crashed. In that situation a fast mode swith to FW would have avoided the crash. 

I've thought about a mode of flight wich takes advantage of wing dynamics for tailsitters. Flying in MC mode a tailsitter I've observed that is so much easy to yaw if you roll at the same time, since wing will turn into wind. What do you think?

AndreasAntener, 2017-02-12:
On the switch problem: I thought about having a 3 position transition switch which would let you do the transition by putting the switch into the middle position, but also lets you switch right through to the other mode by moving the switch all the way. But this also requires you to finish moving the switch all the way after the transition happened.

VTOLDavid, 2017-02-12:
It will be a solution.  From one side to center position will be auto transition and moving from side to side directly, direct switch from one mode to another. Since back transition is always direct you don't need to finish moving the switch.  It could be:
                                                                 Positon_1/Position_2/Position_3
Flying in MC (switch in position_1):   Do nothing/Transition to FW/ Direct switch to FW
Flying in FW (switch in Position_2) :    Transition to MC/Do nothing/Do nothing
Flying in FW (switch in Position_3) :     Transition to MC/Do nothing/Do nothing

But it could be dangerous. If the switch is moved all the way right accidentally (in this diagram) flying in MC it will end in a crash. 

Since transition to FW is a maneuver to gain speed over the stall speed of the wing, transition can be avoided if air speed is over stall. 

It think that my solution will be safer.

closed 2018-02-21 by PX4BuildBot

VTOLDavid, 2018-02-21:
That would be a good enhacement...