PX4/PX4-Autopilot #6573: Snapdragon: power module not yet supported

julianoes, 2017-02-13:
The Snapdragon power module is not yet supported on PX4. Presumably there is an issue because it is on the same I2C bus as the baro.

More information here: https://developer.qualcomm.com/forum/qdn-forums/hardware/snapdragon-flight/30077

mcharleb, 2017-04-13:
@julianoes This is not something we will be able to support on Eagle. Please close the issue unless you think someone else has the bandwidth to work on it, and its a priority.

julianoes, 2017-04-13:
I don't understand this one. How would you get voltage information?

scenkov, 2017-04-16:
I updated the driver for LTC2946 power module in my GitHub repository.

https://github.com/scenkov/DriverFramework/tree/LTC2946
https://github.com/scenkov/Firmware/tree/LTC2946_on_Snapdragon_Flight

julianoes, 2017-04-17:
@scenkov great, thanks. And does it work?

scenkov, 2017-04-18:
Yes it work.

julianoes, 2017-04-19:
@scenkov awesome. It would be great if you could make a pull request with your changes.

limpolito, 2017-06-22:
Hi @scenkov I downloaded your repository and I tried to run your code but I get this error:

[08500/03]  02:37.121  HAP:62:HAP_debug_v2 weak ref not found, return _rtld_sym_zero@_rtld_objmain  0294  symbol.c
[08500/03]  02:37.121  HAP:62:HAP_debug_v2 weak ref not found, return _rtld_sym_zero@_rtld_objmain  0294  symbol.c
[08500/03]  02:37.121  HAP:62:HAP_debug_v2 weak ref not found, return _rtld_sym_zero@_rtld_objmain  0294  symbol.c
[08500/03]  02:37.121  HAP:62:HAP_debug_v2 weak ref not found, return _rtld_sym_zero@_rtld_objmain  0294  symbol.c
[08500/00]  02:37.122  HAP:62:adsp memory mapped   0155  shmem_qurt.c
[08500/00]  02:37.168  HAP:60:In dspal_entry  0220  main.cpp
[08500/00]  02:37.170  HAP:60:adsp memory mapped   0155  shmem_qurt.c
[08500/00]  02:37.175  HAP:54:TEST px4muorb_topic_advertised of [mavlink_log] on remote side... 0167  px4muorb.cpp
[08500/00]  02:37.862  HAP:60:attempting to open the ADSP command file: /dev/fs/px4.config  0184 main.cpp
[08500/00]  02:37.863  HAP:60:reading commands from /dev/fs/px4.config   0191  main.cpp
[08500/00]  02:44.360  HAP:60:7190506 Reset MPU9250   0089  MPU9250.cpp
[08500/00]  02:44.423  HAP:59:7254539 IMU temperature initialized to: 50.091412   0521 MPU9250.cpp
[08500/00]  02:44.425  HAP:60:7256311 additional sensor configuration succeeded   0187  BMP280.cpp
[08500/00]  02:44.426  HAP:60:7257513 LTC2946: Open the device path: /dev/iic-3   0144 LTC2946.cpp
[08500/02]  02:44.426  HAP:60:I2C_Write failed: 285212872 num_bytes_written 0  0414  i2c.c
[08500/00]  02:44.426  HAP:60:7257645 Error: i2c write failed. Reported -1 bytes written   0239 I2CDevObj.cpp
[08500/00]  02:44.426  HAP:60:7257681  1: i2c_write_reg(0x00, &CTRLA, 1)) failed   0112 LTC2946.cpp
[08500/00]  02:44.426  HAP:60:7257712 error: sensor initialization failed, sensor read thread not started   0158  LTC2946.cpp
[08500/03]  02:44.426  HAP:60:Ltc2946 start fail: -5  0132  df_ltc2946_wrapper.cpp
[08500/03]  02:44.426  HAP:60:DfLtc2946Wrapper start failed  0246  df_ltc2946_wrapper.cpp
[08500/02]  02:44.446  HAP:59:Batch transfer failed: 0x110000c8  0524  i2c.c
[08500/00]  02:44.446  HAP:59:7277753 error: read register reports a read of -1 bytes, but attempted to set 6 bytes   0127  I2CDevObj.cpp
[08500/00]  02:44.446  HAP:59:7277794 error: reading I2C bus failed   0253  BMP280.cpp
[08500/02]  02:44.455  HAP:81963:LED: getHandle fail   0280  commander_helper.cpp
[08500/02]  02:44.455  HAP:81963:LED init failed  1362  commander.cpp
[08500/02]  02:44.455  HAP:81963:Buzzer: px4_open fail   0132  commander_helper.cpp
[08500/02]  02:44.455  HAP:81963:Buzzer init failed  1366  commander.cpp
[08500/02]  02:44.456  HAP:81963:Preflight checks always pass on Snapdragon.  0534  PreflightCheck.cpp
[08500/02]  02:44.466  HAP:59:Batch transfer failed: 0x110000c8  0524  i2c.c
[08500/00]  02:44.466  HAP:59:7297688 error: read register reports a read of -1 bytes, but attempted to set 6 bytes   0127  I2CDevObj.cpp
[08500/00]  02:44.466  HAP:59:7297728 error: reading I2C bus failed   0253  BMP280.cpp

Then I disabled the bmp280 because both of them use iic3 and I got another error:

[08500/03]  04:25.053  HAP:62:HAP_debug_v2 weak ref not found, return _rtld_sym_zero@_rtld_objmain  0294  symbol.c
[08500/03]  04:25.055  HAP:62:HAP_debug_v2 weak ref not found, return _rtld_sym_zero@_rtld_objmain  0294  symbol.c
[08500/03]  04:25.055  HAP:62:HAP_debug_v2 weak ref not found, return _rtld_sym_zero@_rtld_objmain  0294  symbol.c
[08500/00]  04:25.055  HAP:62:adsp memory mapped   0155  shmem_qurt.c
[08500/00]  04:25.098  HAP:60:In dspal_entry  0220  main.cpp
[08500/00]  04:25.098  HAP:60:adsp memory mapped   0155  shmem_qurt.c
[08500/00]  04:25.100  HAP:4151:TEST px4muorb_topic_advertised of [mavlink_log] on remote side...  0167  px4muorb.cpp
[08500/00]  04:25.720  HAP:60:attempting to open the ADSP command file: /dev/fs/px4.config  0184  main.cpp
[08500/00]  04:25.720  HAP:60:reading commands from /dev/fs/px4.config   0191  main.cpp
[08500/00]  04:32.370  HAP:60:7271113 Reset MPU9250   0089  MPU9250.cpp
[08500/00]  04:32.433  HAP:59:7335154 IMU temperature initialized to: 54.905815   0521 MPU9250.cpp
[08500/00]  04:32.435  HAP:60:7335757 LTC2946: Open the device path: /dev/iic-3   0144 LTC2946.cpp
[08500/02]  04:32.435  HAP:60:I2C_Write failed: 285212872 num_bytes_written 0  0414  i2c.c
[08500/00]  04:32.435  HAP:60:7335942 Error: i2c write failed. Reported -1 bytes written   0239 I2CDevObj.cpp
[08500/00]  04:32.435  HAP:60:7335978  1: i2c_write_reg(0x00, &CTRLA, 1)) failed  0112 LTC2946.cpp
[08500/00]  04:32.435  HAP:60:7336007 error: sensor initialization failed, sensor read thread not started   0158  LTC2946.cpp
[08500/03]  04:32.435  HAP:60:Ltc2946 start fail: -5  0132  df_ltc2946_wrapper.cpp
[08500/03]  04:32.435  HAP:60:DfLtc2946Wrapper start failed  0246  df_ltc2946_wrapper.cpp

I also tried this fix https://github.com/PX4/DriverFramework/pull/120 but the problem is still there.
Do you know what could be the preoblem?

scenkov, 2017-06-25:
> I also tried this fix PX4/DriverFramework#120

To run LTC2946 you have to boot Snapdragon with battery 2-4s

limpolito, 2017-06-26:
Hi thank you for reply, I'm actually using a 3S battery connected to the APM.

ChristophTobler, 2017-07-06:
What's the current status here?

julianoes, 2017-07-06:
@ChristophTobler I think you would have to check what has been done in these branches and carry over the changes:

> https://github.com/scenkov/DriverFramework/tree/LTC2946
> https://github.com/scenkov/Firmware/tree/LTC2946_on_Snapdragon_Flight

ChristophTobler, 2017-07-10:
@scenkov I have rebased both of your branches (see https://github.com/ChristophTobler/DriverFramework/tree/LTC2946 and https://github.com/ChristophTobler/Firmware/tree/LTC2946_on_Snapdragon_Flight) and tried on my Snapdragon Flight with a 3S battery attached. I couldn't see the voltage.
The [mini-dm output](http://pasted.co/22d21bff) shows some errors which are not present when running master.

closed 2018-02-21 by PX4BuildBot

TSC21, 2018-02-21:
@ChristophTobler what is the status of the power module on the SDFlight?