PX4/PX4-Autopilot #6576: Attitude problem because of changing magnetic field?

nitehack, 2017-02-14:
I have a problem. I am using a pixhawk with the latest firmware with lidar lite v3 and px4flow. I have problem with attitude because I am flying  indoors and there are many changes of magnetic field.
I have tried to change the EKF2_MAG_TYPE to "magnetic heading" but I see that the attitude (pitch, roll and yaw) is continually changing if the magnetic field changes. If I change EKF2_MAG_TYPE to "None", the pitch, roll and yaw attittude is zero, and I dont have feedback of attitude.
I have tried with another pixhawk and the same thing happens.
What is the problem? How to solve it?
Greetings and thanks

tumbili, 2017-02-15:
@nitehack You should post a log, otherwise it's going to be hard to find out what your problem is.

nitehack, 2017-02-15:
@tumbili 
Log with mag type "none" http://logs.uaventure.com/view/kmeAFw4qMYwfQqfgEiuSxE
Log with mag type "magnetic heading" http://logs.uaventure.com/view/Y5kFTpykjc5eAbUHaMK7ti

To download:
https://drive.google.com/open?id=0BxwqX0HkT8MwN0hjdkcyYmxrWFU
https://drive.google.com/open?id=0BxwqX0HkT8MwYXAyamlPTjd3Sms

In the log with mag type "none" I cant see anything , in QGC the attitude is 0.
In the test with mag type "magnetic heading" I didnt fly really because it can be dangerous so I moved the drone and then I put it in the table. Time later you can see that attitude (roll, pitch and yaw) change and compass  rotates. (the drone was remain still) 
[image]

nitehack, 2017-02-17:
Does the pixhawk fuse the local position NED data with the gyro and accelerometer sensors to calculate the attitude?
I have observed that having the drone still and fooling the optical flow, pitch and roll change their value.
Is there any way to disable this fusion?

kd0aij, 2017-02-17:
Try using attitude_estimator_q instead of EKF. Its update() method is quite straightforward.

nitehack, 2017-02-20:
@kd0aij Is it posible to use attitude_estimator_q for attitude estimator and EKF2 for position estimator? I have others problems when I use LPE estimator for local position.

mhkabir, 2017-02-20:
What's the issue with LPE?

nitehack, 2017-02-20:
@mhkabir With LPE for any reason the local position X or Y NED is increasing or decreasing (meters) when drone is stop (I dont know the reason) , with EKF2 this doesnt happend. Futhermore I need that the height control only depends of Lidar lite (local position Z in LPE I see that depends of barometer and Lidar) and I cant select in LPE, with EKF2 I can select only Lidar lite.

kd0aij, 2017-02-20:
@nitehack No, it's either EKF2 or LPE/Q; EKF2 does both position and attitude estimation

tops4u, 2017-02-21:
I assume you do not have a GPS or a secondary Compass which contradicts the other Compass (Orientation wise)? I had this behaviour when I had two Compass (internal and external) with incorrect Orientation settings.

nitehack, 2017-02-22:
@tops4u  I use only the internal compass but I also tried to connect a external compass with GPS, but in both cases is the problem. I observed that when I disconnet px4flow the problem disappears, and when the local position X or Y, also it change the attitude.

tumbili, 2017-02-22:
@nitehack Sorry for the delay, did not realize you posted the logs. I can have a look.

tumbili, 2017-02-22:
@nitehack From the log it seems that you are putting the drone upside down at some points. The result is that the range finder measurements will not be fused anymore and your height starts to drift off. As soon as you turn the drone around again the estimator re-initialises as supposed.
What are you trying to achieve when you have the drone upside down?

nitehack, 2017-02-22:
@tumbili I was testing extreme positions to reproduce the problem. Anyway I have seen that sometimes the problem appears even when the drone is stopped from the beginning (whithout the drone upside down). If I disconnect the optical flow, the problem disappears. Is EKF2 using optical flow data or local position to fuse them with attitude?

tumbili, 2017-02-22:
@nitehack Right, it looks like your magnetometer is disturbed quite a bit (heading innovations are huge). That won't help the optical flow since it needs to convert body frame measurements into earth frame.
Are you sure that your magnetometer is calibrated and that you have set the correct rotation? The only thing that remains to check for me is if the reset of position and velocity states works correctly. The EKF is fusing velocity data from the optical flow sensor, using also a distance to the ground estimate using the range finder.

tumbili, 2017-02-22:
@nitehack I see that this PR https://github.com/PX4/ecl/pull/234 has been merged just recently and brings in some optical flow improvements. Can you test with that?

mhkabir, 2017-02-22:
@tumbili, I think master doesn't have the required submodule update yet.

tumbili, 2017-02-22:
@mhkabir Thanks, I'm doing that now then.

nitehack, 2017-02-23:
@tumbili I have calibrated the magnetometer many times and the orientation is correct. I will test these improvements. Thanks.

marv3, 2017-02-24:
Sorry guys, I'm an idiot. Is it true that lidar lite v3 is supported in the 1.6.0 Release RC1?
And how can I test if the lidar is working?

Sorry for the intrusion hehe

Thanks!!

dagar, 2017-04-11:
@nitehack Any update?

nitehack, 2017-04-16:
@dagar We are building a safe place for the drone before doing the flight test,  so I still can not confirm if the optical flow works well.

alexbennett, 2017-07-25:
I am having this same problem on v1.6.3 and up. Have you found a fix, @nitehack? I have tried LPE and find EKF2 to be far more reliable when it's working so I would prefer to stick with it. This issue seems to randomly pop up after a varying number of flights.

closed 2018-02-21 by PX4BuildBot