PX4/PX4-Autopilot #6596: sdlog2 python script usage is wrong.

GregoireH, 2017-02-16:


DaveSprague, 2017-03-19:
Agreed, just spend an hour trying to figure out why the sdlog2_dump.py script wasn't working at all.

It's line 306 in the code where it specifies the separater between MSG name and the first FIELD name as "_"

I'm still not getting any actual data in the csv file output but at least now I get the header line.

Attached is the .BIN file I'm trying to convert to a csv

closed 2017-03-20 by LorenzMeier

bellahduan, 2017-10-10:
@DaveSprague  hi, Sorry to disturb you . but i am confused . the log download from QGC is ***.px4log . but it can not use to analysis . Is there any way to solve the problem?
thank u so much .

bellahduan, 2017-10-10:
@DaveSprague 
[image]