PX4/PX4-Autopilot #6602: SITL stops with negative time

julianoes, 2017-02-17:
I've run into the case with jMAVSim SITL on Mac where it suddenly says:

```
ERROR [platforms__posix__px4_layer] WARNING! TIME IS NEGATIVE! -849055500 vs -847483835
WARN  [navigator] global position timeout
```
The simulation is unusable at that point.

patoMuerto, 2017-04-18:
I am encountering this error as well.

I am running Ubuntu in VirtualBox on a Mac host.  After reading this issue (https://github.com/PX4/Firmware/issues/5703) and the link describing TSC, I forced the sitl to run on a single cpu by adding taskset to sitl_run.sh.  Below is the echo of the appended command.

```bash
 SITL COMMAND: taskset -c 0 \ 
      /home/dev/px4/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4 \ 
      /home/dev/px4/src/Firmware \
      /home/dev/px4/src/Firmware/posix-configs/SITL/init/ekf2/iris
```
I have not fully tested this but I have not seen the error since.

LorenzMeier, 2017-08-20:
Should be fixed in recent versions.

closed 2017-08-20 by LorenzMeier

docHell, 2019-03-05:
It is still the same 

ERROR [Unknown] WARNING! TIME IS NEGATIVE! 140078086 vs 140078087

Every time this error appear I lost the udp connection.
How can I solve ?

julianoes, 2019-03-07:
@docHell please create a new issue. Add as much console output as possible (in  ```), describe your version, your choice of simulator,  etc. thanks.