PX4/PX4-Autopilot #6611: Old style or new style flight mode assignment

AndreasAntener, 2017-02-18:
@LorenzMeier @dagar @DonLakeFlyer 
I've added 2 new switches for the old style flight mode assignment in https://github.com/PX4/Firmware/pull/6567:
- Manual switch
- Stabilized switch

The previous state was that we used Acro in fixed-wing as stabilized switch. This is now solved but it will need docs and UI changes to reflect that.

My first basic question is if we should even keep that switch assignment method or if we should let it go and focus on ironing out the problems with the new style. I do like the old style because allows me to use 2 switches without any RC mixing. Am I missing something here?

dagar, 2017-02-18:
Thinking out loud, could we extend the new style to achieve what you want? An optional second switch that the firmware would be responsible for mixing with the first to extend the number of mode slots?

tubeme, 2017-02-19:
I think it is tooo much switches already. With the VTOL it is even one more. So isn't it better if we just follow the logic of flying.

So I will give an example with the VTOL where the confusion is worse.

With the "acro switch" programmed I can change between Manual - Acro (in quad mode) and Manual - Stabalized (in airplane mode). It is very easy to be one switch. We can use the acro switch but give it another name. Like "Manual" switch instead of acro switch.

The thing is that it could be a 3 pos switch so we could give one more mode:

Manual           - Acro              - Ratitude (MC Mode)
Stabalized      - Manual          - Acro (FW mode)

1000us             1500us             2000us

That is the way the switch should work in between the two modes. This way when we use the VTOL independent from the transition switch we will have consistent and expected behavior from our machine. The third mode in MC and FW mode could be assignable. But the first 2 or rather 4 should be as is. 

That is it plain and simple!!! Another switch is killing me.

There is another thing that is annoying. I cannot start Pause from all modes. I have to switch to mission and then to pause in order to activate it. This is sometimes very cumbersome for the user. He is hitting the loiter switch and seeing in awe that the plain is not loitering. Until i hit him the mode switch to mission.

tubeme, 2017-02-19:
If the "Manual switch" is not programmed by default it should be in Manual mode for MC and Stabilized in FW.

We can change the position of Manual, Acro in the 3 pos switch but they should be linked to corresponding Stabalized and Manual for FW.

AndreasAntener, 2017-02-19:
@dagar yeah I could imagine that one would set a secondary channel. For each slot on the main channel you would have the possibility to specify "use secondary modes". When that is selected we would store another 6 secondary slots for that main slot. That would give you 6x6 possible combinations.

Example:
- main 1: use secondary
  - sec 1: stabilized
  - sec 2: acro
  - sec 3: manual
- main 2: alt control
- main 3: pos control

tubeme, 2017-02-19:
I try to simplify my system and I got this config so far.

Left Hand - all concerning flying

SE - Mode switch
    1. Manual
    2. Assisted
    3. Mission

SA - Assisted switch
   1. Altitude
   2. Position

SB - Manual switch
   1. Manual (MC) / Stabalized (FW)
   2. Acro (MC) / Manual (FW)

For VTOL

SF - Transition switch
   1. MC
   2. FW

Right Hand - Emergency + Take Off and Landing

SC - Emergency switch
     1. Normal flight
     2. Auto Pause
     3. Return

SD - Landing gear and Flaps
     1. Normal Flight
     2. Flaps
     3. Further flaps + Landing gear

SG - Throttle CUT for the Gas engines.

SH - OSD switch.

I have no switches left for anithing more...

tubeme, 2017-02-19:
Left hand and right hand logic is simple. During flight mode change your hands mandatory should be on the controls of Ail and Elev. With your left hand you can leave the Throttle and Rudder and do the change while in full control over the flight dynamics of the plane. While in the other situations - emergency, landing, taking off you can leave the Ail and Elev controls and activate the needed.

dagar, 2017-03-09:
To summarize and plan a path forward. If we added an optional modifier switch which unlocks a 2nd set of mode slots would that be good enough to drop the old style setup and concentrate everyone's testing and development effort?

fredowski, 2017-07-26:
Hi, thanks for the pixhawk firmware! I can give some feedback from absolute beginners. We tried to configure the flight modes via QGC. The "single" switch method worked pretty much out of the box. I could assign "Manual", "Stabilized" and "Mission" to the three switch positions on my Taranis. Yesterday we tried to use two 3 level switches to have one more mode "Altitude". We thought that it should be pretty simple but it turned out that it is even impossible to have "Stabilized" and "Altitude". There is this "automagic" logic in QGC which will assign the "Assist" and "Auto" Modes to the "Mode" switch depending on wether you select certain flight modes. 

This is really difficult to understand when you do not know the logic behind this. Today I looked in the firmware and figured out all modes are somewhat assumed to be single switches and that there is this priority logic to select the actual mode. Plus the 3 level "Mode" switch. 

Without reading anything we assumed that it works like the "use secondary mode" proposed above, then we thought we were just too stupid to figure out the right combination of channel assignment and thresholds...

So for me it would be intuitive to have this option in the "single switch" scenario to use a secondary switch to actually select the flight mode.

We have assigned the Armed/Disarmed and Kill to dedicated switches.

dagar, 2017-07-28:
Thanks for the feedback @fredowski. I think it's important that we limit the feature set to one well supported (and documented) way of doing things.

closed 2018-02-21 by PX4BuildBot