PX4/PX4-Autopilot #6612: Not able to set Magic Mode for indoor fake gps data

rs1990, 2017-02-19:
Hello,

I have been trying to set up my system to be able to send vicon data to my pixhawk, but i cannot find and  change SYS_COMPANION to 257600 and INAV_DISAB_MOCAP to 1 (disable mocap estimation) in the parameter list. i am able to everything else. the pixhawk is working fine with the companion computer and i am able to arm and disarm via ros commands. 

Is there something i need to change and recompile the firmware ?. i am using the latest released firmware.

Please help. This has been an issue for quite a while.

Thank you,

RS.

rs1990, 2017-03-07:
Hello,

Id appreciate any help. 

Thanks.

julianoes, 2017-03-08:
The param `INAV_DISAB_MOCAB` won't exist or work because the inav position controller is no longer built and used.

The estimator that you can now select are ekf2 and lpe. I'm not familiar how they interface to Vicon position data I'm afraid.

karanchawla, 2017-03-16:
@julianoes Are you saying that I can't do indoor navigation using inav anymore? The problem I'm facing is that though I have my `att_mocap_msg` (sent via a small udp script I wrote) I can't get the mode to switch to position control.

mhkabir, 2017-03-17:
@ChristophTobler We need to either remove the Fake GPS Devguide page or update it on how to get it working with new estimators. It is very confusing for users, and also isn't clearly distinguished from the standard VICON code path.

ChristophTobler, 2017-03-17:
@mhkabir Yes you're right. I'll ask @vilhjalmur89. I think he used it recently.

ChristophTobler, 2017-03-17:
@rs1990 We haven't used it in a while and currently don't have the possibility to test it ourselves but here's how it should work:
There is no distortion and INAV anymore. So proceed as in https://dev.px4.io/advanced-fake-gps.html with the following changes
* in step 1, just do the first part. Ignore the `launch_fake_gps_distorted.sh`
* in step 3, ignore the distorted part
* skip steps 4 and 5
* steps 6-9 should be the same

Let us know if it works

dagar, 2017-04-12:
@rs1990 Please reopen if this isn't resolved.

closed 2017-04-12 by dagar