PX4/PX4-Autopilot #6620: MC: Won't start after arming（Firmware: release 1.4.4）

DeanJJJ, 2017-02-20:
Similar to #4083, nothing happened while pushing Throttle up after arming by safety switch and RC. At this moment QGC shows servo1_raw to servo6_raw are all 900, normally they should be 1075. PX4 release 1.4.4 was running.

kd0aij, 2017-02-20:
Have you tried a later release?

DeanJJJ, 2017-02-20:
@kd0aij no, I thought it was fixed in issue #4083 but now it appears again

kd0aij, 2017-02-20:
Maybe it was fixed on master only?

DeanJJJ, 2017-02-21:
@kd0aij there is something different, the problem I met not looks like related to non-manual modes, I doubt it might be caused by time-clock but can not prove it.

kd0aij, 2017-02-21:
If master doesn't work, please post a log

DeanJJJ, 2017-02-21:
@kd0aij issue #4083 was fixed in April 2016 and release 1.4.4 was published in August 2016, it should be included in that release. This problem does not appear very frequently, around once in 15 attempts.
I will post a log later.

mhkabir, 2017-02-21:
As we keep saying, USE THE LATEST RELEASE.

DeanJJJ, 2017-02-23:
@mhkabir I have tried to use firmware release 1.5.5, but the drone was totally out of control which is similar with issue #6070. Thank you for help.

DeanJJJ, 2017-02-23:
@mhkabir I have tried to use firmware release 1.5.5, but the drone was totally out of control with HITL simulation in jMAVSim which is similar with issue #6070. Thank you for help.

mhkabir, 2017-02-23:
Use LPE for now as the issue says.

DeanJJJ, 2017-02-23:
@mhkabir I have tried to change efk2 to LPE previously but it was still out of control.

LorenzMeier, 2017-02-23:
@DeanJJJ Unless you provide a log file we will unable to help you as there is no information to go on. I will close this issue tomorrow unless we get additional details.

We have a dedicated test flight team performing daily test flights, so I'm confident that whatever you are experiencing is somewhat related to your setup. That's why we need additional information.

DeanJJJ, 2017-02-24:
log file attached for the data of crashing with HITL simulation in jMAVSim.
[crash_log.zip.gz](https://github.com/PX4/Firmware/files/798516/crash_log.zip.gz)
the issue of "Won't start after arming" does not appear very frequently so it needs time to make a recurrence. I will upload log file if it happens again.

dagar, 2017-04-12:
@DeanJJJ any update?

dagar, 2017-04-12:
Please reopen with a log if you're able to reproduce.

closed 2017-04-12 by dagar