PX4/PX4-Autopilot #6630: Mavlink stream rate & connection with QGC

Youngdong-Clark-Son, 2017-02-22:
Hello all.
 
I recently bought Pixhawk Mini. Currently, I am setting the hardware to establish ROS-Pixhawk-GCS system.

In the previous Pixhawk it provided TELEM2 port which can be connected to the ROS-computer directly. However, Mini provides only one TELEM1 port which is assigned for Wifi-module now. The last port I can use is the port with 'Android'? cable used when bootloading. I successfully increased the stream rate of that port by adding a file to the SD card in /etc/extras.txt.

The content is as follows :
mavlink stream -d /dev/ttyACM0 -s HIGHRES_IMU -r 200
mavlink stream -d /dev/ttyACM0 -s LOCAL_POSITION_NED -r 200
mavlink stream -d /dev/ttyACM0 -s ATTITUDE_QUATERNION -r 200

With that file, Mini now provides high frequency data about 170hz with the 'Android' port.

However, when I try to connect Mini from QGC, it shows no response at all although I connect my wifi well with low ping.

When I remove the extras.txt file again, QGC can communicate with Mini very well.

1. Any idea how I can change the stream rate without affecting the Wifi module(TELEM1)?
2. As I know, The 'Android' port is assigned with ttyS0 and the TELEM1 port is ttyS1. But all people I saw use ttyACM0. When I tried to use ttyS0 instead of ACM0 in extras.txt file, it failed.

Thank you for time.

dagar, 2017-04-11:
Hi @Clark-Youngdong-Son, were you able to resolve this?

LorenzMeier, 2017-04-30:
@Clark-Youngdong-Son You need to differentiate between the port on your host PC and on the autopilot. It sounds like your issue is coming from using instructions for one side on the other, which yields non-surprisingly inconsistent results.

closed 2017-04-30 by LorenzMeier