PX4/PX4-Autopilot #6631: Gazebo HIL

L4ncelot, 2017-02-22:
Hi,

I've been using px4 firmware for some time with ROS and Gazebo simulator in **SIL** configuration. 

Now what I would like to do is to test my algorithms in **HIL** configuration. The idea is to take data from simulated sensors in Gazebo and send them to my px4 which would send flying commands back to Gazebo. 

We're using **navio2** board with **raspberry PI**. I was thinking about connecting **GCS**, in which the simulation runs, with my **navio2** board through USB. Then modify **gazebo_mavlink_interface** plugin from **sitl_gazebo** package to send messages through **serial port** instead of **UDP**. And then somehow modify px4 firmware on board to accept those messages.

Or is there another way? I saw someone doing **HIL** with **jMAVSim** but I have no idea how to make this work with Gazebo.

Thank you for any advice.

TSC21, 2017-07-25:
@L4ncelot this is ongoing with https://github.com/mavlink/mavros/pull/684. Then, we just need HIL interface working, which `rotors_simulator` already have, though not using this plugin.

L4ncelot, 2017-07-26:
@TSC21, well I've figured it out a long time ago. I've basically connected the **navio2** board with my computer through ethernet and ran the same ROS instance on both devices. Then I just edited the **UDP** link in gazebo model of iris copter to my **navio2** IP address and on the **navio2** board ran other components we needed to test alongside SITL simulation to be able to control copter in gazebo. It's not true HIL, because of the running simulation on the board, but it gave us the clue which components of our code could be run on the board and which couldn't.

Or at least I think I did it this way, can't remember, because we're using something else right now.

closed 2017-07-26 by L4ncelot

TSC21, 2017-07-26:
>  It's not true HIL, because of the running simulation on the board, but it gave us the clue which components of our code could be run on the board and which couldn't.

Actually it's not but if it fit your needs, good. Still, the HIL<->ROS interface is already possible through the above plugin. Right now we just need proper Gazebo support.

iftahnaf, 2020-04-13:
> @TSC21, well I've figured it out a long time ago. I've basically connected the **navio2** board with my computer through ethernet and ran the same ROS instance on both devices. Then I just edited [...]

Do you remember which parameter you changed in the iris.sdf?

I'm trying to do the same, running the roslaunch mavros px4.launch on my RPi 3b+. 
I typed export ROS_MASTER_URI=http://RPiAddress:11311 on both Rpi and my PC.
Then, no_sim=1 make px4_sitl gazebo and last roslaunch gazebo_ros.
I can see the mavros topics from the Pi and from the PC, but there is no connection between them: In the /mavros/state the connection status is false.

L4ncelot, 2020-04-13:
@iftahnaf is your ROS networking working properly? I don't have my work computer with me right now, but there are some variables for the **mavlink_interface** gazebo plugin you can play with found in the model.

iftahnaf, 2020-04-13:
Thanks for the answer, Ill look for that.
The ROS networks seems to be fine, I can subscribe to mavros topics from both Pi and PC.
I launch mavros from the Pi, with roslaunch mavros px4.launch, with the fcu_url definition:
fcu_url:="udp://:14540@PCAddress:14557". is it ok?
Thanks again :-)

L4ncelot, 2020-04-13:
It depends on your configuration... the `fcu_link `should be connected to the autopilot binary. So if you're running autopilot's binary on the PI, set the IP to the **PI_address**. Look at the px4 documentation explaining all the variables etc. It will help you a lot in understanding the configuration.

You need to check the configuration yourself. I'm using the system differently now, so I don't have the default values in my head anymore.

iftahnaf, 2020-04-13:
The autopilot's binary is on the PC and I set the fcu_link to it.
 I will check the documentation and my configurations again.
Thank you.