PX4/PX4-Autopilot #6636: krait update param failed AFTER upgrading BSP image and aDSP image [Snapdragon Flight]

gpanders, 2017-02-22:
This only started occurring in the last day or two.

Upon starting PX4, I get an error message saying:

```
ERROR [muorb] Initialize Error calling the uorb fastrpc initialize method...
ERROR [muorb] ERROR: FastRpcWrapper Not Initialized
```

followed by a long list of `error finding param: VT_B_TRANS_DUR`.

The mini-dm output repeats the following:
```
HAP:52:undefined PLT symbol dm_read (1004) /libpx4.so 0303 symbol.c
HAP:52:undefined PLT symbol dm_write (652) /libpx4.so 0303 symbol.c
```

I used the latest BSP image (Flight 3.1.2) and ADSP images from the Intrinsyc website (both were last updated 11/30/2016). I have reflashed the firmware on both the board and DSP, as well as completely deleted by Firmware directory, re-cloning it, rebuilding, and reuploading. Nothing works.

LorenzMeier, 2017-02-22:
@jywilson @mcharleb @bharath374 Do you have a chance to look into this?

genemerewether, 2017-02-24:
This is a problem with commander linking - disabling commander in the QURT build or commenting out the lines around the dm_read allow the code to run.

dm_read is used at the following line:
https://github.com/PX4/Firmware/blob/058bc97c838d9ff3958e95c3ad18098f2d75ef36/src/modules/commander/commander.cpp#L1475

ChristophTobler, 2017-03-06:
I get the same error with latest master

ChristophTobler, 2017-03-06:
I now get 
```
px4 starting.

INFO  [sdlog2] [blackbox] /root/log/2017-03-06
INFO  [sdlog2] [blackbox] recording: 08_48_12.px4log
INFO  [dataman] Unkown restart, data manager file '/home/linaro/dataman' size is 11797680 bytes
error finding param: VT_B_TRANS_DUR
error finding param: FW_CLMBOUT_DIFF
error finding param: VT_B_TRANS_DUR
error finding param: VT_B_TRANS_DUR
error finding param: VT_B_TRANS_DUR
error finding param: VT_B_TRANS_DUR
error finding param: VT_B_TRANS_DUR
error finding param: VT_B_TRANS_DUR
error finding param: VT_B_TRANS_DUR
error finding param: VT_B_TRANS_DUR
error finding param: VT_B_TRANS_DUR
error finding param: FW_THR_CRUISE
INFO  [mavlink] mode: Normal, data rate: 1000000 B/s on udp port 14556 remote port 14550
Sleeping for 1 s; (1000000 us).
pxh> INFO  [mavlink] using network interface wlan0, IP: 10.20.21.87
INFO  [mavlink] with broadcast IP: 10.20.31.255
INFO  [mavlink] partner IP: 10.20.21.14

pxh> 
pxh> ekf2 start
error finding param: FW_ARSP_MODE
pxh> INFO  [lib__ecl] EKF aligned, (pressure height, IMU buf: 21, OBS buf: 16)
WARN  [lib__ecl] EKF baro hgt timeout - reset to baro
WARN  [lib__ecl] EKF baro hgt timeout - reset to baro
```

And the IMU data is completely off. This is with the snapdragon laying flat on my table.
[image]

@jywilson Any ideas? Or am I missing something?

gpanders, 2017-03-06:
@ChristophTobler are you seeing any side effects from commenting out that block? Or is the issue you posted about with the IMU data the result of commenting?

ChristophTobler, 2017-03-07:
@gpanders Not sure if the weird IMU data is the result of this. Without commenting, I just get the `HEARTBEAT` topic

ChristophTobler, 2017-03-07:
@jywilson here's the mini-dm output when the lines commented in the commander.cpp: http://pasted.co/75a23e3b

And here's the output with current master:
http://pasted.co/89730c99

Hope it helps. Thanks!

jywilson, 2017-03-07:
Thanks Christoph.  I am looking at this now.

jywilson, 2017-03-07:
From the mini-dm output it looks like the dm_write and dm_read symbols were added to the libpx4.so (aDSP binary) build.  These functions are not defined at link time *and* are not defined when the .so is loaded so the loader in the aDSP fails the loading of the libpx4.so (the flight stack).  The AppsProc portion of the flight stack continues to run, but none of the FastRPC function calls into the aDSP from the AppsProc will work.

Essentially this is a build issue.  I will continue looking and update you shortly.

Jim W.

genemerewether, 2017-03-08:
@jywilson - the undefined symbols come from src/modules/dataman, which isn't included in the QURT build (and didn't build when I briefly tried). See my comment above for a quick fix.

jywilson, 2017-03-08:
@genemerewether - the dataman functions are implemented as stubs in the px4_qurt_impl.cpp file.  The dm_read and dm_write functions were in fact being linked in, but were being redefined as C++ functions since the function signatures were different than that defined in dataman.h.  The function signature of dm_read and dm_write was changed recently in dataman.h (and dataman.c), but were never changed in px4_qurt_impl.cpp.

Thanks to @mcharleb for helping me spot this.

I have pushed a fix to the master branch.  The PR is here:
https://github.com/PX4/Firmware/pull/6748

Jim W.

ChristophTobler, 2017-03-08:
I can confirm that https://github.com/PX4/Firmware/pull/6748 fixed the `krait update param failed` issue. However, I still have some IMU problems. see https://github.com/PX4/Firmware/issues/6749

closed 2017-03-08 by ChristophTobler

mhkabir, 2017-03-08:
I'm just curious here - Why can't we just run commander on the apps processor and avoid those stubs?

julianoes, 2017-03-08:
> I'm just curious here - Why can't we just run commander on the apps processor and avoid those stubs?

I would agree. Isn't that already the case?

mhkabir, 2017-03-08:
Nope : https://github.com/PX4/Firmware/blob/master/posix-configs/eagle/flight/px4.config#L14

It's running on the aDSP.

julianoes, 2017-03-08:
You're right, nevermind.

jywilson, 2017-03-08:
The reason the commander is running on the aDSP is to avoid the overhead of the many uORB messages that would need to move between the aDSP and AppsProc.  Commander subscribes to just about everything and running on the AppsProc would probably have a performance impact.  

Also, the last time we tried this, I think we had problems with certain uORB functions which are not supported in the multi-uORB module (muORB).  muORB is responsible for mapping uORB transactions between processor cores on the Snapdragon SoC.  

It would be interesting to try this again, just to see if the same issues exist, but I am not sure it's the best alternative from an architectural perspective.

julianoes, 2017-03-08:
@jywilson ok makes sense. Thanks for the explanation.