PX4/PX4-Autopilot #6641: Non-satisfactory flight test results

LorenzMeier, 2017-02-22:
@MaEtUgR @Stifael Some of the recently merged work got negative feedback from our QA team:
http://review.px4.io/plot_app?log=f72bdea3-c111-422c-99e6-8d8ab3cb1b0d

They were testing this:

[image]

santiago3dr, 2017-02-22:
couple flights; good performance
http://review.px4.io/plot_app?log=6ef60043-b047-4ecb-88f8-d908b4f8daf8
http://review.px4.io/plot_app?log=060c9780-1d50-46e2-a5ef-a89e7195cb0f

LorenzMeier, 2017-02-22:
@tumbili The flight tests suggest the EKF performance degraded. Any ideas?

tumbili, 2017-02-23:
@LorenzMeier The updates from ecl were about range sensor and optical flow fusion so I don't expect this to be related.

MaEtUgR, 2017-02-23:
My change simply allows to detect landed with a stick position that is not 5% from the bottom but 15%. If I remember right it was already like this at some time in the past. Now you even have a parameter (LNDMC_MAN_DWNTHR) to tweak or disable this on the newest master.

Stifael, 2017-02-28:
can we close this?

closed 2017-03-23 by Stifael