PX4/PX4-Autopilot #6646: How to use Vicon for local position estimation

jay3ss, 2017-02-23:
I am using the PX4, MAVROS, and a Vicon system and am attempting to use the Vicon to obtain the position of a quadcopter. I have followed the instructions given in the [External Position Estimation tutorial](https://dev.px4.io/external-position.html) but for some reason I am getting the differing values for the local_position/pose and the mocap/pose (as shown in the figure below). Also, the local_position/pose drifts substantially throughout time. 

I am using the PX4 firmware from approximately 01/26/17 (sorry, I can't find how to check the firmware version). Here are the parameters that I set:
- ATT_EXT_HDG_M = 2 (motion capture)
- LPE_FUSION = 12 (only fuse vision position and fuse vision yaw checked)
- LPE_VIC_P = 0.0000 (I set 0.0001 but it changed to this value instead)

What I've done to work around this is to "throw away" the messages on the local_position/pose topic by remapping them to another topic. Then, I publish the pose given by the Vicon along the /mavros/mocap/pose and /mavros/local_position/pose topics. This seems to work OK, but is not ideal. Am I missing some parameters that I need to set? I'm at a loss right now as to how correct this or how exactly how I'm supposed to use the Vicon for local positioning.

[image]

nahush8, 2017-02-23:
I have been working on the same problem for quite a while now. Check issue: #6555 . My LPE_FUSION parameter is 4.  I am getting the exact same pose on both topics though when I try to fly a quadrotor in position control mode, it drifts. and is not stable.

bhairavmehta95, 2017-03-15:
@nahush8 did you ever find a solution to this?

nahush8, 2017-03-15:
@bhairavmehta95  Check https://github.com/PX4/Firmware/issues/6555 .. my mocap and local estimates matched perfectly. I still couldn't get the Quadrotor to fly stably in either position control or offboard mode.

bhairavmehta95, 2017-03-16:
@nahush8 are you using the 1.4.4 Release like was suggested here https://github.com/mavlink/mavros/issues/591 ?

Are were you still using the latest px4fmu_lpe?

nahush8, 2017-03-16:
@bhairavmehta95  I used the latest code and loaded px4fmu_lpe firmware.

mhkabir, 2017-04-02:
Closing since no logs.

closed 2017-04-02 by mhkabir

abdullahmohiuddin, 2017-05-28:
@nahush8 were you able to fly stably ?
I think in my case, the LPE pose and the mocap pose are the same, but still there is some unstability.

nahush8, 2017-05-28:
@abdullahmohiuddin  Nope ! I could never get it running perfectly. 
Try Fake GPS using Mocap: https://dev.px4.io/en/advanced/fake_gps.html 
It works much better than this.

TSC21, 2017-05-28:
@jgoppert any thoughts on this?

abdullahmohiuddin, 2017-05-29:
@nahush8 Have you tried Fake GPS?
Could it be because when you fake GPS, you keep fusing the other sensors?

abdullahmohiuddin, 2017-05-29:
[image]
[image]
[image]

These are my x y z  setpoints and also the way it follows it. What do you think the problem could be?

mhkabir, 2017-05-29:
You need controller tuning.

abdullahmohiuddin, 2017-05-29:
Thank you so much @mhkabir for the earliest reply. 
It seems in my case that the attitude rate and attitude control, both are working fine. 
I suspect the velocity control needs tuning. How do you suggest I should move forward?
I mean , am I correct in assuming my velocity control needs tuning?

[image]

mhkabir, 2017-05-29:
>am I correct in assuming my velocity control needs tuning?

Yes, seems so.

abdullahmohiuddin, 2017-05-29:
Ok, is there any recommendation, as to where to start, and how to proceed?
Should I increase the parameters and with what percentage?

crvogt, 2017-06-19:
Hi @jay3ss 
I was wondering, did you modify the vicon_bridge code to advertise vicon pose data on the mavros/mocap/pose topic? The pixhawk documentation says mavros provides the data by default (as suggested by the mocap use-pose parameter being set to "true" in the px4_config.yaml file) but can't figure out how mavros is supposed to be getting the pose data, ie, where is it listening to the vicon provided topics?
thanks for any help!

jay3ss, 2017-06-20:
@crvogt, I ended up remapping the `mavros/local_position/pose` topic to something else then republished the vicon data along that  topic and `mavros/mocap/pose`. Since I did that I haven't had any problems.

crvogt, 2017-06-20:
Okay, it sounds like what you posted originally effectively. As you mentioned though, it doesn't seem ideal, like there's a built in way that mavros can simply listen to the vicon topics. But if it works, I'll probably try it after I get frustrated enough :) Thanks for the help!

crvogt, 2017-06-20:
Hi @mhkabir , I don't suppose you could clarify the best way to go about incorporating the vicon data into mavros? I assume mavros is looking for a mocap topic, but unsure of where that's happening.
Thanks!

crvogt, 2017-06-21:
For future reference, mavros will listen directly to Vicon and incorporate it almost out of the box. The only thing that needs to be changed is the topic in the callbacks in "mocap_pose_estimator.cpp".
Thanks again for the input.

Seanmatthews, 2017-07-25:
Jumping on this thread because the question is relevant: How does one use only position (and not orientation) from Vicon for local position estimation? I'm told there's a setting to ignore orientation from the mocap topic, but it's not apparent that that's the case.

mhkabir, 2017-07-25:
@Seanmatthews You can set ATT_EXT_HDG_M to 0, which will disable vision yaw fusion.

Seanmatthews, 2017-07-25:
@mhkabir My setup is such that I'm providing a vision_pose_estimate that includes pose, and a mocap_pose_estimate that includes only position. If I set ATT_EXT_HDG_M to 0, then it won't calculate heading from vision_pose_estimate data, right? I'm not sure that's what I want?

YuYangguang, 2017-11-05:
As the latest version of firmware had removed the LPE module as well as the related parameters, did you try to use mocap as the external position with EKF2 module? I don't know how to use the mocap funtion in the firmware after v1.5.2? Can you help  me? @mhkabir @crvogt

TSC21, 2017-11-05:
Removed LPE?

TSC21, 2017-11-05:
@YuYangguang https://github.com/PX4/Firmware/tree/master/src/modules/local_position_estimator. What setup are you using (MoCap system, OBC, flight controller=?

YuYangguang, 2017-11-10:
I'm really sorry to reply you late, I'm really busy recent days. 

It's my fault that I did't describe this issue clearly.   Now I want to use Mocap system to control the mini UAV in vicon environment.  In the last months, I use the version v1.5.1 dev of PX4 firmware as the controller. But now I want to update my firmware to the latest. But when I finished update, problems comes.   

Firstly, I noticed that  the log in the git of v1.5.2 stable write, of which the commit ID is  b0ee5256d5ea0f40fe4e3dded9be27bbc3ffa871,  as follows:

 Disable LPE in px4fmu-v2_default
    With GCC 4.9 the binary is to large for the flash memory.
    This is why we disabled LPE on that platform.

More importantly, the parameter ATT_EXT_HDG_M was removed, namely it cannot found in QGC.  In the previous version, I set it as motion capture, so I can dump vicon data into the topic "/mavros/mocap/pose". But now since this parameter was removed, I did't know how to set parameters to make the PX4 receive the vicon data.

In my last comment, I said that LPE was removed. I also mean that it could not be found in QGC anymore. I can only see the EKF parameters.

Now I just want to seek help in how to set the PX4 parameters to make the PX4 receive the vicon data. Then I can get the local position of uav in vicon  environment.       I don't know whether I describe this issue clearly enough.  If you can give me some tips, I will really appreciate it. Thanks

TSC21, 2017-11-10:
https://dev.px4.io/en/advanced/switching_state_estimators.html

YuYangguang, 2017-11-11:
Thank you for your kind reply @TSC21 . 

But this guide document is only valid for version 1.5.2 or older version. Even when I followed the guide of the document, I still could not found the parameter ATT_EXT_HDG_M in the QGC.  Could you please explain it more accurately? Thank you for your kindness.

TSC21, 2017-11-11:
This guide works for any version above 1.5.2 also. If it did not worked we wouldn't have it available.