PX4/PX4-Autopilot #6651: Takeoff not detected after 2nd time in altitude mode

western-co-at, 2017-02-23:
Hello
With release v1.6.0rc1 or the current master the takeoff seems not to be detected after the second time arming in altitude mode:
http://review.px4.io/plot_app?log=b29f4633-e7c9-49a9-ab92-18106d9f01f9

```
2017_02_23_09_31_15: [logger] file: /fs/microsd/log/2017-02-23/09_31_1
2017_02_23_09_34_09: ARMED by RC
2017_02_23_09_34_11: Takeoff detected
2017_02_23_09_34_29: Landing detected
2017_02_23_09_34_31: DISARMED by auto disarm on land
2017_02_23_09_34_36: ARMED by RC
2017_02_23_09_34_46: DISARMED by auto disarm on land
```
The first time arming in altitude, the takeoff and landing is detected fine.
After the 2nd time arming, the motors are only spinning with idle thrust for some seconds and the attitude/position controller is not responding to any rc commands or nick/roll movement.
In case I switch to e.g. stabilized mode and then back to altitude mode again (when the motors are spinning with idle thrust) the takeoff is detected, the motors are spinning up and the attitude/position controller seems to be working again.

western-co-at, 2017-03-02:
I did some furthermore tests in this issue. The problem is only present when the local position estimator hasn't already set the xy_valid flag. This will only be set when a gps signal is available. So in case you are flying indoor the flag won't be set...

For a short test I uncommented the following line:
https://github.com/PX4/Firmware/blob/e892a51afb59763255d48d5df2613c1e10b713bc/src/modules/land_detector/MulticopterLandDetector.cpp#L314
The MC is now starting normaly in altitude mode without gps - also after the n times.
But I found the following comment:
https://github.com/PX4/Firmware/blob/e892a51afb59763255d48d5df2613c1e10b713bc/src/modules/land_detector/MulticopterLandDetector.cpp#L314
...so I think there has to be done something more instead of uncommenting this line.

xn365, 2017-03-12:
Encounter the same problem.

LorenzMeier, 2017-03-12:
@western-co-at Can you try to set up your throttle ranges correctly as described here?
https://dev.px4.io/land-detector.html

In addition: LPE is a developer setup and you might need to dive into the code to ensure it all works. We are maintaining EKF2 right now as the standard community / user version which gets tested. Land detection on master and with EKF2 is well tested and reliable.

western-co-at, 2017-03-15:
Tried now with current master, the default settings for the landing detector and without gps signal. The problem still exist.

There was a misinterpretion on my part: I meaned the vehicle_local_position uOrb message instead of the LPE. 

I think the issue #6818 describes the same bug.

Stifael, 2017-04-06:
@western-co-at  can you not just upload a log?

Stifael, 2017-04-07:
I tried to reproduce in sitl, but not succesful yet.

western-co-at, 2017-04-07:
I have already posted a Log File in my first comment: http://review.px4.io/plot_app?log=b29f4633-e7c9-49a9-ab92-18106d9f01f9
If you need more log files with different test cases - please let me know. With a real flight controller it was reproducible for me after the 2nd time & without gps signal.

Stifael, 2017-04-07:
@western-co-at: I am not sure why this comment is even there (https://github.com/PX4/Firmware/blob/e892a51afb59763255d48d5df2613c1e10b713bc/src/modules/land_detector/MulticopterLandDetector.cpp#L188), since the vertical velocity is anyway extracted from the imu.

Stifael, 2017-04-07:
@western-co-at: did you do auto takeoff or manual takeoff? a log file would be appreciated (preferably ulog)

western-co-at, 2017-04-07:
@Stifael: here another short log with current master firmware: http://review.px4.io/plot_app?log=8136102d-90e0-43f9-a6e7-b1e05161a30a
...tested again without props and without movement of the copter.
I have done these tests without auto takeoff. Please also have a look at the Raw Radio Control Inputs and the corresponding Actuator Outputs (Ouput 0 - Output 3).

Stifael, 2017-04-11:
@western-co-at: can you please tell me what LNDMC_MAN_DWNTHR parameter is set to (I cant see it from the log since this parameter is unfortunately not used before logging starts.
The LNDMC_MAN_DWNTH defines the stick threshold for thrust. once exceeded, the vehicle should not detect landed anymore. From the plot below, however, you can see that it says in landed despite having stick input large.
On the other hand that still does not explain why the problem only occurs after the first takeoff.

western-co-at, 2017-04-11:
The LNDMC_MAN_DWNTHR parameter was set to 0.4 during the last log. But also when I set this parameter to the default value (0.15) it behaves the same.
Again: When I uncomment this line https://github.com/PX4/Firmware/blob/39dc983c6f4554e4825b31a595e250b040cdaa8b/src/modules/land_detector/MulticopterLandDetector.cpp#L314 its working without gps signal. Here the according log (again with LNDMC_MAN_DWNTHR set to 0.15): http://review.px4.io/plot_app?log=ae36e698-66d4-40d0-9626-17ade88ac718 
But I don't think that this is the solution for the problem...

MaEtUgR, 2017-04-12:
There might be an issue with the logic in the case you don't have GPS that went under my radar because even when testing this stuff in altitude mode I was always outside and had GPS.

https://github.com/PX4/Firmware/issues/6651#issuecomment-283575626 might have spotted the problem: you don't need a valid xy world frame position/velocity to detect verticalMovement (still concidering you have a brometer). I have to test this but altering the condition like suggested might be a viable solution.

Stifael, 2017-04-12:
@western-co-at, I just think the problem is not the valid_xy flag. you should exit the landdetector once the sticks are above the LNDMC_MAN_DWNTHR, otherwise there is another bug.

Stifael, 2017-04-12:
https://github.com/PX4/Firmware/blob/master/src/modules/land_detector/MulticopterLandDetector.cpp#L171-L172
_has_manual_control_present()  should be true;
_manual.z < _params.manual_stick_down_threshold should be false -> hence it should be false
 manual_control_idle || !_control_mode.flag_control_manual_enabled should also be false assuming !_control_mode.flag_control_manual is not set incorrectly 
Therefore, manual_control_idle_or_auto should be false and as a result of that landdetection should be false. The vehicle should not even enter _has_position_lock().
Its a bummer that I could not find the control_mode in the log, because it could be that this log is not set correctly

Stifael, 2017-04-12:
ok I got it:
it enters this line:
https://github.com/PX4/Firmware/blob/master/src/modules/land_detector/MulticopterLandDetector.cpp#L228, which returns true and therefore you stay landed.

western-co-at, 2017-04-13:
@Stifael: Thanks #7029 solved the problem in this specific case. I have tested this without props again. I have not tested this in flight if this has other unpredictable side effects.

May I close this issue?

Stifael, 2017-04-13:
yep, you may close it. you always can reopen it

closed 2017-04-13 by mhkabir

MaEtUgR, 2017-05-03:
should be fixed by https://github.com/PX4/Firmware/pull/7081