PX4/PX4-Autopilot #6660: Param not stored to flash mem

sanderux, 2017-02-24:
We crashed one of our prototypes today due to a parameter that didnt store.
The param is question was PWM_AUX2_REV, the following happened;

A new FC as mounted and a bench test was done.
Elevons needed reversing so they were set through QGC.
After setting a bench test was done on the elevons and they performed as required.
Then we went to the field and started a transition, the plane crashed right after transition.
Post-mortem analysis indicated PWM_AUX1_REV was set to 1 but PWM_AUX2_REV was at 0
We managed to test the params on the original wings and indeed the response was incorrect.

The deduced bug is that the param was loaded in memory but it failed to write to flash. QGC showed no warning upon storing the param.

[image]

davids5, 2017-02-24:
@sanderux - what board config is this on?

sanderux, 2017-02-24:
Dropix px4-fmu-v2

davids5, 2017-02-24:
@sanderux,

Just to to confirm does the HW has a RAMTROM FRAM on it and is the config the px4fmu-v2_default build?

I ask because there are "flash based param" in other HW configs and the title may need to be changed to help guide in debugging this.

sanderux, 2017-02-24:
Yeah i'm not sure. I'll check. It is the fmuv2 default build
But the bug is more a request to verify the param stored in whatever storage it resides

sanderux, 2017-02-25:
@davids5  I cant find the specs for the type of storage.
It;s this one: https://drotek.com/shop/en/drotek-parts/478-dropix-flight-controller.html

kd0aij, 2017-02-25:
FM25V01 FRAM: https://github.com/drotek/Dropix/blob/master/Hardware/Dropix.PDF

davids5, 2017-02-25:
@kd0aij - thanks for the print

LorenzMeier, 2017-02-25:
There is also the possibility that the param did not get transmitted, potentially because of an input validation issue in the UI.

sanderux, 2017-02-25:
I am 100 percent sure the params did get applied before reboot as i bench tested the elevon responses both through transmitter and stabilize responses.

closed 2017-04-18 by sanderux